hoglet67/RGBtoHDMI #320: Palette for Apple ][ wrong?

JPT77, 2023-03-24:
Hi, 

RGBtoHDMI works fine with my Apple II, except for the color palette.

calibrating the two color pots at J14 and F14 only made the R2H lose sync, it did not change the color at all.
So, I was able to remove a little noise but not fixing the colors.

see screenshot:
https://mega.nz/file/SNhmmAiJ#byOo3IAM_FqQixnz0xiefSuFsWKIq_Tz501ee_x1Rcg

do you have any idea what could be the problem?

btw, there is no difference if I use PAL or NTSC quartz. Either way R2H detects 60 Hz input.

thanks

timmcross, 2023-03-24:
I have an Apple II Plus I would like to connect to a RGB2HDMI.  Is this a II+ or a IIe?  I already have several IIe's outfitted with this device.  I just can't find the cable diagram for a II+.

JPT77, 2023-03-24:
It's an Apple IIeuroplus which is a II+ except b/w PAL. I modified it to NTSC, so I get a colored image.
Cable diagram? 
https://github.com/hoglet67/RGBtoHDMI/wiki/Cables
it's just pin 4 Video, everything else GND.

IanSB, 2023-03-24:
@JPT77 

You have to set the chroma phase quadrant (one of four possible options will be correct for your hardware).

Either:
Press SW3 to cycle through the four phase quadrants
or
Change the Artifact Phase in the palette menu

Save your configuration when complete

JPT77, 2023-05-10:
Works fine, thanks.

closed 2023-05-10 by JPT77