hoglet67/RGBtoHDMI #321: Support for older Analog boards?

hoglet67, 2023-03-27:
Hi Ian,

I have a couple of V2 analog boards that I would like to bring back into use.

I'm currently testing with an RGBtoHDMI issue 2, the BBC CPLD v7.9, the BBC Profile and the latest Beta56.

I've hit a couple of issues.

**Issue 1: The analog board is incorrectly detected as a V3 board this can't be changed manually.**

Looking at the code, it seems that it's not possible to differentiate between the following boards:
```
   FRONTEND_ANALOG_ISSUE3_5259
   FRONTEND_ANALOG_ISSUE2_5259
   FRONTEND_ANALOG_ISSUE1_UA1
   FRONTEND_ANALOG_ISSUE1_UB1
```
That's fine, except there is no longer a way to manually set the interface, because the interface setting is now hidden due to:
```
 HIDE_INTERFACE_SETTING
```
Would it be possible to decide at run time whether to hide the interface setting, based for example on whether the frontend info returns a range of possible interfaces?

For now, I've commented HIDE_INTERFACE_SETTING in defs.h so I can continue testing.

**Issue 2: The green clamping is not working on the V2 Analog board with the latest BBC CPLD**

What's happening is the CLAMP signal on pin 4 of the P2 connector is always ow:
```
        clamp_int <= not(csync1 or csync2) and sp_data;
        analog <= 'Z' when version = '0' else clamp_int;
```
In this case it's sp_data (GPIO0) that low.

Looking a cpld_rgb.c I think this is happening because of the supports_bbc_9_12 flag.
```
          switch(config->rate) {
                      default:
              case RGB_RATE_3:
              case RGB_RATE_6:
                    if (supports_bbc_9_12) {
                        RPI_SetGpioValue(SP_DATA_PIN, 0);   //no ac coupling
                    } else {
                        RPI_SetGpioValue(SP_DATA_PIN, coupling);
                    }
                    break;
```
So it looks like V 7.8 of the BBC CPLD uses SP_DATA to enable 9/12 bit capture modes rather than controlling AC Coupling?

Is this all expected? 

I've got a couple of workarounds:

- Revert to a version of the BBC CPLD older than 7.8
- Switch to the RGB CPLD and use the Electron profile (as the BBC profiles are not present)

What's the main use case for the 9 and 12 bit modes with the BBC CPLD? VideoNuLA?

Dave

IanSB, 2023-03-28:
@hoglet67 

As mentioned above, if you add the following to the bottom of config.txt replacing any existing #interface values, that should permanently set the configuration to an Issue 2 board on a standard software install and it should show in the system summary menu as issue 2.
(Interface_3 is YUV and Interface_C is BBC / RGB)

#interface_3=12
#interface_C=6

(The numbers might change if any further interface designs are added)

IanSB, 2023-03-28:
>What's the reason the clamp output of the CPLD needs to be enabled/disabled under software control? What would break if it were enabled all the time?

On the issue 3 boards the clamp is permanently connected to the video path via C1 irrespective of the setting of the AC/DC switch U6 and I recall it introduced a noticeable disturbance in the DC level of the green/Y channel when DC coupled if the source was high impedance.