hoglet67/RGBtoHDMI #323: 128 pixel high screen

sjgray, 2023-05-12:
In the geometry menu for "Min V Height" the hardcoded minimum is 150. Is there some reason for this limit? I am trying to emulate an old vintage LCD panel that has a resolution of 480x128 (80x16 text)  using a Commodore VDC chip but this 150 pixel limit means I can't get the video to fill the screen. Can it be lowered?

IanSB, 2023-05-15:
@sjgray 

That value has already been reduced to 100 in recent betas, see this commit:

https://github.com/IanSB/RGBtoHDMI/commit/05203862679ac5209d514ccdcfd6e9b77355da85

Latest beta here:
https://github.com/IanSB/RGBtoHDMI/releases

BTW A new stable release is imminent which will also have that change.

On a sidenote, I'd like to get a profile created for the colour output of the Superboard 600D but I still haven't built mine.
I recall from the osiweb forums that you have been experimenting with the 600D colour output so if I create a speculative profile would you be able to try it as I assume your post means you have an RGBtoHDMI.

sjgray, 2023-05-15:
That’s great news! Looking forward to trying the beta tomorrow and then the release.
Yes, I have a few OSI C4P and super board machines and my own colour board. I’d be happy to test for you.

closed 2023-05-15 by sjgray

sjgray, 2023-05-15:
The beta worked great, thanks!

IanSB, 2023-05-17:
@sjgray 

The stable release is now available:
https://github.com/hoglet67/RGBtoHDMI/releases

sjgray, 2023-05-17:
Nice, thanks!