hoglet67/RGBtoHDMI #328: How to switch between different saved configs?

quadrupedal, 2023-06-10:
How can I have two or more saved configs and easily switch between them?

I can see the "Saved Config" option in the menu, and can change that to "Alt 1" then make changes before saving the configuration, but those settings seem to be applied to all the configs. Changing to a different saved configuration doesn't give me the changes I've made.

My use case is that I have a monitor that can do 1600x1200 at 60Hz only, but will do 50Hz at 800x600, so I thought I would make a "productivity" config with the slightly nicer higher resolution, and a "gaming" config of the lower resolution (scaled up again by the monitor) but at 50Hz for smoothness.

Is this possible?

If it is, is there a way to switch between configs without going through the menu? Basically, once everything is set up, I'd like to use a single button to switch between modes and never see the menu again.

Thanks!

quadrupedal, 2023-06-13:
Ok, thanks. Could we make this a feature request? I'm sure the same issue would affect people who have different monitors, projectors, etc. that they want to switch between without opening up their computer.

In the meantime I'll look into SD card switchers. If one will physically fit then I use that as a workaround, even though having to power down makes it less than ideal.

The monitor in question is a Philips 200P4 (200P4MG/ 00 to be exact), https://www.p4c.philips.com/files/2/200p4mg_00/200p4mg_00_pss_.pdf, which I'm running through a mini HMDI to DVI-D cable. The manual states 800x600@56Hz, but it will go down to 50 in that mode. In 1600x1200 mode it displays a "Cannot display this video mode" error when you do the 50Hz test. It feels like it could do it if it wanted, but someone decided not to let it.

IanSB, 2023-06-13:
@quadrupedal 
>Could we make this a feature request? 

OK if I can think of an easy way of implementing such a feature.

>The manual states 800x600@56Hz, but it will go down to 50 in that mode. In 1600x1200 mode it displays a "Cannot display this video mode" error when you do the 50Hz test. It feels like it could do it if it wanted, but someone decided not to let it.

Yes, it's a shame more older monitors don't support 50Hz as there is usually no technical reason for such a limitation. One 1600x1200 monitor that fully supports 50Hz with smooth motion is the HP LP2065. There is also the Dell 2007FP from the same era which does display 50hz sources but it does that by doing an internal 50hz to 60Hz conversion so you end up with judder just the same as using 60Hz directly so it is essentially useless for smooth motion etc.

Newer monitors are generally better at supporting 50Hz because they usually support 50Hz HD resolutions but even that isn't guaranteed to work.