hoglet67/RGBtoHDMI #329: Apple IIe Taxan RGB Aux slot video

timmcross, 2023-06-17:
Hello, still loving this product.  I have built so many of these for my retro computers.

My latest is a Taxan RGB adapter for the Apple IIe.  I believe it does TTL 3-bit color with composite sync.  When I connected it to the RGB2HDMI, the screen was rolling on the CGA profile.  I connected the CSYNC to the HSYNC pin.  I found a stable image using the Acorn Atom profile.  The colors are close and I assume they won't be correct since there is no intensity bit.

Could a closer profile be created for it?  I took some screenshots and gathered some log files.  Let me know what else would help.
[image]\
[Taxan RGB.zip](https://github.com/hoglet67/RGBtoHDMI/files/11777765/Taxan.RGB.zip)

IanSB, 2023-06-17:
@timmcross 

Looking at the logs, you don't appear to be using the latest stable release which has a changed profile folder structure so I suggest you update to that first
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20230517_eb620884

Then try the following:
[Taxanprofile.zip](https://github.com/hoglet67/RGBtoHDMI/files/11778891/Taxanprofile.zip)

Unzip the above and put in \Profiles\6-12_BIT_RGB\Apple

Use the new select profile menu to select the profile.

This still has the palette set to RGB which has a limit of 8 colours against the 16 of the standard Apple palette so the colour choice will be restricted unless the output is actually analog RGB, not TTL RGB.

With Apple RGBI monitors, the 16 RGBI values were mapped to a different palette set than CGA and that more closely resembled the apple colour palette but I don't have any details on how the Taxan monitor handled that.

There is a Laser 128 profile which follows the Apple RGBI palette

You need to find out some more info on this setup.
i.e:
Is the output TTL or analog?
If it is only 3 bit TTL did the TAXAN monitor map those 8 levels to different on screen colours from the standard 8 RGB values.
(Might have had a switch to change palette)

You could create an alternative palette to map the 8 values to different colours but as the Apple II palette is 16 colours it is always going to be a compromise.

timmcross, 2023-06-18:
Thanks for that!  I setup the new files on a formatted SD card and tried it with the Taxan profile and the Laser128 palette.
Here is a shot of one of the RGB lines, I appears to be TTL to me but let me know.
[image]

I also found a separate line with activity on the card so I hooked it up as the intensity line.  Here is a color test.  I think this looks correct.
[image]

The text is green because of a setting on the card.  I think I can set it to amber, blue, or white as well.  Let me know if you want some other logs or screen shots from this setup.  Thanks again!

IanSB, 2023-06-18:
@timmcross 

Looks good, I think the correct palette to use should be "RGBI_(XRGB-NTSC)"
XRGB is what Apple called their version of RGBI with the different palette on the Apple III
(The Laser palette is identical anyway but probably better to use the one with the official Apple name)

Do you know if this board works on 50Hz Apple IIe?

If everything looks OK can you run an auto calibration and post your saved profile and I'll include that in the next release.

IanSB, 2023-06-18:
@timmcross 

Also post your wiring details for inclusion in the wiki

timmcross, 2023-06-18:
Sorry, all of my IIe's are 60Hz.  Is it possible to do a PAL mod to an Apple IIe (thinking not...).

And for sake of completeness, my card is technically labelled as a Panasonic Video-7 card yet someone on a forum pointed out to me that it is a replica of the Taxan card.  I can't find any documentation on the Panasonic one I have.
[image]
[Taxan410-80a.pdf](https://github.com/hoglet67/RGBtoHDMI/files/11782103/Taxan410-80a.pdf)

Here is the pinout:

Card   -   RGB2HDMI CGA adap
1     -     1    GND
2     -     4    Green
3     -     8    Comp Sync
5     -     3    Red
9     -     5    Blue
10   -     6    Intensity

After switching to XRGB, here is a screenshot of the color bars.  Seems that it is a little off.
[image]

IanSB, 2023-06-18:
@timmcross 
>After switching to XRGB, here is a screenshot of the color bars. Seems that it is a little off.

Yes, I just realised the colour are mapped differently so stick with the laser palette for now (Although I might rename that to something more generic in future)

I'll include a 50Hz profile in case it does work with such apple IIe variants.

closed 2023-12-27 by timmcross