hoglet67/RGBtoHDMI #330: Wyse WY-700: No Sync Detected

A-Small-Mice, 2023-06-18:
[image]
[image]
[image]

I have encountered an issue with the Wyse WY-700 when using RGBtoHDMI. It seems that RGBtoHDMI is unable to properly analyze the signals emitted by the Wyse WY-700, resulting in only a gray screen being displayed. I attempted to address this problem by changing the Clock Frequency of RGBtoHDMI to 50MHz, as the Wyse WY-700 uses a 50MHz Xtal. However, this change did not yield any noticeable difference. Are there any other suggestions or potential solutions I can try?

IanSB, 2023-06-18:
@A-Small-Mice 

Post a screencap of the source summary page which shows the timing info detected.
(press SW2 + SW3 to screencap in that page)

A-Small-Mice, 2023-06-21:
Below are the images of the screen and information that appear when using Wyse-700 as a CGA card
[image]

[image]
When using Wyse-700 and Hercules simultaneously on a single computer, I have confirmed that the Wyse-700 is recognized as a secondary adapter in the Hercules screen, as shown in the above image. The 'mode co80' command also functions properly.

IanSB, 2023-06-21:
@A-Small-Mice 

The source summary shows no sync being detected so either the board is not working / not initialised or the sync is on different pins or at different voltage levels like 1v p-p.

Firstly do you have the second board that plugs into the header as you only show one in the photo?

Do have any technical info on the pinout of the 9 way connector?

I found a manual on archive.org that mentions 4 levels of grey which is not possible with a standard CGA/ EGA connector so the pinout may be completely non standard.

Also it's likely that this board outputs the high resolution signal under all circumstances and just appears to be CGA etc from a software point of view, translating such modes to the high resolution signal as that would make using a dedicated hi res monitor easier and the manual I found only mentions the following hi res timing for the monitor, not standard CGA etc:

[image]

I think it should be possible to capture a signal with such timing but I need to know exactly what's on that 9 way connector.

If you don't have any further technical info, do you have an oscilloscope to probe the pins to look for sync signals etc?

Maybe taking a photo of the underside of the 9 way connector would show which pins had pcb tracks connected to them as there don't seem to be any on the top side (unless this is a 4 layer board)

A-Small-Mice, 2023-06-22:
@IanSB 

[image]
[image]
[image]
[image]
[image]

Firstly do you have the second board that plugs into the header as you only show one in the photo?
>Yes.

Do have any technical info on the pinout of the 9 way connector?
>No, but I used a multimeter to confirm that pins 8 and 9 of the Wyse-700 card connector are GND.

Also it's likely that this board outputs the high resolution signal under all circumstances and just appears to be CGA etc from a software point of view.
>This card is both a CGA card and a Hercules card. It has jumper settings.
>I have seen a screen displayed in Hercules mode using a Chinese scaler known as the XVGA box. However, I am not sure how the scaler was able to properly display the screen despite its pins 8 and 9 were ground (GND). Unfortunately, I do not currently have the XVGA box to reconfirm.

If you don't have any further technical info, do you have an oscilloscope to probe the pins to look for sync signals etc?
>Unfortunately, I don't have an oscilloscope. However, I can provide some information based on measurements I obtained using a multimeter.

**Wyse-700 Pinout?**
Pin 9 GND
Pin 8 GND
Pin 7 No Connection?
Pin 6 Connected Somewhere
Pin 5 No Connection?
Pin 4 Connected Somewhere
Pin 3 Connected Somewhere
Pin 2 Connected Somewhere
Pin 1 Connected Somewhere

IanSB, 2023-06-22:
@A-Small-Mice 
>No, but I used a multimeter to confirm that pins 8 and 9 of the Wyse-700 card connector are GND.

If pins 8 & 9 (H & V sync) are really grounded then this is definitely a non standard connector. They might have done that deliberately as feeding the wrong frequency syncs into some monitors can damage them.

>This card is both a CGA card and a Hercules card. It has jumper settings.

It still could output only high resolution video by upscaling CGA/Hercules frame buffers with the jumper settings merely determining how the card appears to software. The apparent non standard pinout seems to support this hypothesis.

>I have seen a screen displayed in Hercules mode using a Chinese scaler known as the XVGA box. However, I am not sure how the scaler was able to properly display the screen despite its pins 8 and 9 were ground (GND). Unfortunately, I do not currently have the XVGA box to reconfirm.

They could have made up a custom cable after working out the pinout.

I think you really need to know the pinout before this can progress further.

Anyway here is a profile based on the monitor timings from the Wyse manual above but you may have to make a custom cable, assuming the board actually works.
It expects 4 bits of video on the standard CGA RGBI pins and sync on the standard CGA H & V sync pins
(This will not be 4 grey levels but if/when you get something displayed this can be improved by using different palettes etc)

[Wyse_WY-700.txt](https://github.com/hoglet67/RGBtoHDMI/files/11840572/Wyse_WY-700.txt)

Put it in 
\Saved_Profiles\6-12_BIT_RGB\IBM PC Compatible
(You must use the latest stable release as the folders have changed)

Some suggestions:

Try the above profile (just in case it works)

Get someone with an oscilloscope to look at it.

Use one of the cheap 8 channel PC based logic analysers available on ebay to capture the digital output of all the pins to see what is happening

This is the type of logic analyser I mean:
https://www.ebay.co.uk/itm/314272582941
(There may be local suppliers where you are)

Try connecting the green channel to each of the pins you identified (+GND) and see what sort of rolling image you get on screen as that might identify the sync lines by the pattern they make.

Try connecting the H & V sync signals in all combination to the 5 pins you think are used (assuming only +5v on those pins, not 12v etc) This would be 20 combinations I think and one should result in sync detected with the timings matching the above monitor specs (i.e. 69.4Hz with line timing of ~31250ns in the source summary menu

A-Small-Mice, 2023-06-23:
**IN CGA MODE, WY-700 PROFILE**
try 1
RGBtoHDMI Pin 8(Purple) -> WY-700 Pin 2
RGBtoHDMI pin 9( Grey ) -> WY-700 Pin 1

[image]
[image]

Result: Vertical green lines captured 

try 2
RGBtoHDMI Pin 8(Purple) -> WY-700 Pin 1
RGBtoHDMI pin 9( Grey ) -> WY-700 Pin 2

Result: A red vertical line swaying on the right side of the sreen 

Src=70Hz
Disp=60Hz

[image]
[image]

I think that pins 1 and 2 of the WY-700 are related to the sync because the other pins are not responsive. Attempts to connect pins 1, 2, and 3 of the RGBtoHDMI to other pins of the WY-700 were not successful.

**IN HERCULES MODE, MDA PROFILE**
RGBtoHDMI Pin 8(Purple) -> WY-700 Pin 1
RGBtoHDMI pin 9( Grey ) -> WY-700 Pin 2

[image]

Src=70Hz
Disp=60Hz

Attempts to connect pins 6, 7 of the RGBtoHDMI to other pins of the WY-700 were not successful.

IanSB, 2023-06-23:
@A-Small-Mice 
>try 2
RGBtoHDMI Pin 8(Purple) -> WY-700 Pin 1
RGBtoHDMI pin 9( Grey ) -> WY-700 Pin 2
Result: A red vertical line swaying on the right side of the sreen
Src=70Hz
Disp=60Hz

OK it looks like you've found the sync lines as the values in the source summary for line duration and frame rate are as in the manual but the results also show that the profile is not correct in other areas.

Here is a corrected profile which has the line length set to 1600 and the lines per frame set to 448 which I think should be correct based on the readback of 69.67hz and 32005ns :

[Wyse_WY-700.txt](https://github.com/hoglet67/RGBtoHDMI/files/11850755/Wyse_WY-700.txt)

However you got it working mostly by luck as you don't appear to have connected a ground line as one is not shown in the photos and that will cause all sorts of instability.
As mentioned above you need to connect a ground line from GND on the RGBtoHDMI (pin 1) to GND on the Wyse board

>Attempts to connect pins 6, 7 of the RGBtoHDMI to other pins of the WY-700 were not successful.

Pins 6 & 7 are intensity or secondary Green and secondary Blue so would not work
You need to connect to the primary RGB lines which are 3, 4, 5 of the RGBtoHDMI connector

So try the following:
Install the updated profile (delete any old saved copy in /Saved_Profiles as that will override the update in /Profiles)
Connect the two sync lines as above that gave you Src=70Hz
Connect the GND line from Pin 1 of the RGBtoHDMI to a GND pin on the Wyse
Connecting Pins 3, 4, 5 of the RGBtoHDMI  (RGB) to the three unknown pins on the Wyse

The source summary page should show
Clock Error close to 0 (less than 4500)
Line Duration: approx 32000
Lines per frame 448
Frame rate 70Hz (approx 69.75Hz)

Also check the above when changing the link from CGA to MDA.
If they don't change then it is certain that the card always outputs the above timing and scales by pixel replication the CGA/Hercules buffer to the higher resolution (1280x800 interlaced)

Post another screencap of the source summary page when done

A-Small-Mice, 2023-06-24:
[image]
[image]
[image]

Unfortunately, it seems that luck hasn't been on my side. I tried the method you mentioned.

Now I can only obtain a white screen. Pins 6 and 7 of the WY-700 seem to have a special purpose, but they are unresponsive. Its Pins 1, 2, and 3 appear to be constantly held at 1.

IanSB, 2023-06-24:
@A-Small-Mice 
>Unfortunately, it seems that luck hasn't been on my side. I tried the method you mentioned

It looks like the timing for the profile is now setup correctly and the image would be captured correctly if the right video signals were available so you have correctly identified the sync signals.

There are a few possibilities:
1. The video signals are not at TTL levels and the voltage levels are lower such as 1v rather than 5v. There are no driver transistors shown on the photos but there are some low value resistors that may be acting as voltage dividers to reduce the voltage levels.
Try tracing the pcb tracks of unknown pins back to where they are connected both visually and using your meter to see if they are connected to those resistors. If so try connecting the RGB inputs of RGBtoHDMI to the other side of those resistors.
(Maybe confirm the other side of the resistors are connected to one of the TTL chips and not ground or +5v)

2. Maybe the board is faulty in some way
Are you sure the PC is booting correctly? e.g. do you get disk activity if you type dir after booting etc.

IanSB, 2023-06-24:
@A-Small-Mice 

This is possibly how the resistors are arranged:
[image]

I notice there are two groups of low value resistors R12, R13, R14 and R29, R30, R31 and both of them might be arranged like the above if they are connected to the 9 pin Dsub in which case you would have to trace back through two resistors to find the TTL chip pin.