hoglet67/RGBtoHDMI #334: Support HP 9000/300?

JPT77, 2023-07-10:
Hi, 

I got an HP9000 300 series and cannot get a video signal. (monitor is missing)
the early models did not have VGA but composite video. someone told me it could be 18khz?
is there a way for RGBtoHDMI to support this signal? 
Should I try to fetch a signal using an oszilloscope?

thanks

hoglet67, 2023-07-10:
There is some useful information here:
https://hp-series300.net/hardware.htm#S7

Can you identify which graphics board you have?

JPT77, 2023-07-10:
Well, none. Just onboard Cinch.
but that's a great site, because it shows much more detail.

hoglet67, 2023-07-10:
What's the HP model number of the CPU/system board?

Or the product number for the machine as a whole?

Maybe you could post a picture or two?

JPT77, 2023-07-10:
With the help of your link I found out that it's the Model 332. With 68030 CPU and no extended gfx. From the back it looks just like the 310.
only addon is a GPIO 98622A, which i believe is some 16 bit parallel bus.
and the 9153C drive case with 2x20MB.

IanSB, 2023-07-11:
@JPT77 
>With the help of your link I found out that it's the Model 332.

According to page 55 of this manual:
http://www.bitsavers.org/pdf/hp/9000_300/98572-90039_Series_300_Model_332_Service_Handbook_Feb89.pdf
the video is either composite mono or RGB with sync on green.
In either case you should be able to get some sort of image using the analog board and an appropriate profile although actual results may depend on number of bits per pixel etc.

>Should I try to fetch a signal using an oszilloscope?

That would be a useful start to estimate the sync timing

JPT77, 2023-07-16:
[config.txt](https://github.com/hoglet67/RGBtoHDMI/files/12063610/config.txt)
[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/12063611/Log.txt)
[image]
[image]
[image]

I tried it today and the "Superboard II 64x32" profile gives the best result. 
it shows some text at two different locations and flashes on and off in about 1 second interval. (see mainmenu image)

do these pics and files help? 

I will fetch an oszi image on tuesday.

IanSB, 2023-07-16:
@JPT77 

Here is an experimental profile to try, the sampling clock and line length are probably wrong but it might give you a locked image:
[HP_9000-332_Test_V1.txt](https://github.com/hoglet67/RGBtoHDMI/files/12064274/HP_9000-332_Test_V1.txt)

First install the latest stable release from here:
https://github.com/hoglet67/RGBtoHDMI/releases
(This has a new select profile option and the folder structure is different)
Then put the above file in:
/Profiles/6-12_BIT_YUV_Analog/_Other

The use the select profile option to pick "Other" then the above profile.

You may get a readable image but the pixel sampling is likely going to be incorrect and you will also have to adjust the H and V offsets to centre the image.

The easiest way to determine the correct pixel clock is to examine the board with the video connector and note the frequencies of the crystals which are usually printed on their cans. (post photos if necessary)

Post a screencap of the source summary page (not a photo)
Press SW2 & SW3 to screencap when displaying that page, file saved on the SD card.

JPT77, 2023-07-18:
Great, You are a composite magician ;)

ive got two crystals 33.333 mhz and 35,904 mhz
i think it's the second because someone told me it was 18khz?!

[image]
[image]

[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/12085291/Log.txt)

IanSB, 2023-07-18:
@JPT77 
>ive got two crystals 33.333 mhz and 35,904 mhz

I used 16.666Mhz as the pixel clock as I could see the 33Mhz crystal on some photos but the second one is almost certainly used for the pixel clock as it divides exactly into the measured line length

Here is a new profiles with 18Mhz clock
[HP_9000_Composite_Test_V2 - 18Mhz.txt](https://github.com/hoglet67/RGBtoHDMI/files/12089306/HP_9000_Composite_Test_V2.-.18Mhz.txt)

There is some possibility that the clock is 36Mhz so if the characters are clean but look uneven and blocky that might be the reason.

As before post screencaps of the source summary page so I can check the timing.

There are other tweaks required such as adjusting the offsets in geometry to centre the image and the DAC levels in sampling to optimise the sync threshold and the video (Yhi) threshold. (You adjust the values up and down until something breaks and then put the value in the middle of the good range)
If you get a clean result, post your saved profile (will be in /saved_profiles)

Also do you know if the video supports bright characters (i.e. an intensity setting) as that will require a different palette and an additional DAC threshold to distinguish between normal and bright video. (I have seen some screenshots that showed bright chars)

JPT77, 2023-07-20:
Capturing 2+3 from menu produces empty files. That's why you didn't get screenshots from menu yet (just realized)

`$ hd capture1.png 
00000000  89 50 4e 47 0d 0a 1a 0a  00 00 00 0d 49 48 44 52  |.PNG........IHDR|
00000010  00 00 03 c0 00 00 00 00  01 00 00 00 00 f2 ad 29  |...............)|
00000020  1f 00 00 00 12 49 44 41  54 78 01 05 c1 81 00 00  |.....IDATx......|
00000030  00 00 00 10 ff d5 08 00  00 00 01 26 9d 27 46 00  |...........&.'F.|
00000040  00 00 00 49 45 4e 44 ae  42 60 82                 |...IEND.B`.|
0000004b
`

We estimated some timings using the oscilloscope:
row freq 25.4 to 25.375 khz
frame 60hz
row sync pulse 2,700us
vertical pulse 120us

I just don't know much about this and had to rely on the "expert" ;)

h offset 72 or 76 looks fine (some screens start a letter earlier than others)
v offset between 3 and 21. It doesn't seem to use the top and bottom most rows, but you cannot be sure exactly.
the screen is most likely 80x25.
sampling phase 0 and 4 look best.
yhi 038-059, set it to 49.
sync threshould - don't understand ;)

picture looks ok now. it's centered, text doesn't flicker that much any more. But the letters are still a little odd, eg the capital letter i is missing the left side of the top and bottom bar. but maybe this is how the font looks like.

IanSB, 2023-07-20:
@JPT77 
>Capturing 2+3 from menu produces empty files. That's why you didn't get screenshots from menu yet (just realized)

I'll have a look at that as I have just reproduced it.

>picture looks ok now. it's centered, text doesn't flicker that much any more. But the letters are still a little odd, eg the capital letter i is missing the left side of the top and bottom bar. but maybe this is how the font looks like.

That's indicative of a 36 Mhz pixel clock rather than 18 which seems likely as the font would then be 14x16 rather than 7x16 which would make the chars look a lot better quality.

Here is a 36Mhz profile:
[HP_9000_Composite_Test_V3 - 36Mhz_.txt](https://github.com/hoglet67/RGBtoHDMI/files/12113578/HP_9000_Composite_Test_V3.-.36Mhz_.txt)

NOTE: This will not work in YUV mode but will work in RGB mode so put the above file in:
/Profiles/6-12_BIT_RGB_Analog/_Other
and reboot
Then select it using the select profile menu which will automatically reprogram the CPLD to RGB mode

Post a photo of the source summary page so I can check the timings

JPT77, 2023-07-21:
[image]

We are lucky, I took a photo of the source summary. 
This is before I did the finetuning.

I cannot continue testing before some time next week.

IanSB, 2023-07-21:
@JPT77 
>We are lucky, I took a photo of the source summary

That looks like the sync timing is correct but the characters are showing signs of undersampling.
 e.g. the ( and ) chars should be mirror images but have a different shape.
This means that the 36 Mhz clock is likely the correct one.
However I have just realised there is another potential problem:
The standard analog board design uses MAX9144 chips which have a bandwidth limit of somewhere in the 25 - 35 Mhz range so might not work with a 36Mhz clock. You would probably get an image but not be able to fully clean up the noise.

You could try changing the MAX9144 chips to MAX9108 chips which have a higher bandwidth (25ns vs 40ns propagation delay) or another alternative would be to get a pre-assembled board as the latest revision of those boards uses a different design with much higher bandwidth chips (6ns propagation delay) which will probably cope with >100Mhz bandwidth

This is what they look like:
Top is github DIY design
Middle is first version of assembled boards (same bandwidth as the DIY design)
Bottom is latest version of assembled boards (with high bandwidth)
[image]

IanSB, 2023-07-26:
@JPT77 

Have you run an auto calibration? 
Also try adjusting the DAC-E:Sync level up or down in the sampling menu and re-running the auto calibration
Similarly try adjusting the DAC-A: G Hi level up and down.

I have a test board that can generate pretty much any video signal that RGBtoHDMI can capture so I programmed it to reproduce the timing of HP9000 composite and I could get an image with both MAX9144 and MAX9108 but they were both noisy even after calibration (MAX9108 was better). The only way I could get a totally clean image was to use one of the new fast design boards.

[image]

This is the mono version of the above that only looks at the green channel:
[image]