hoglet67/RGBtoHDMI #335: Which version do i need?

U32Frank, 2023-07-13:
I have an older RGB2HDMI board, [this ](https://github.com/c0pperdragon/Amiga-Digital-Video/tree/master/direct_adapter) one. 

The latest release just gives me a black screen. Is there a specific version of the software that i need for this board?

IanSB, 2023-07-14:
@U32Frank 

The latest stable release should work with the c0pperdragon board, I just tested it again to confirm that.

Is this a new installation?

If so, either the board is faulty or your have an early revision Amiga 500 or 1000 as some of the signals were missing on those early motherboards and had to be manually wired up.
What Amiga model are you using and what motherboard revision?

U32Frank, 2023-07-14:
I have an Amiga 500 6a

IanSB, 2023-07-14:
@U32Frank 
>I have an Amiga 500 6a

That should work.

A few other things to try:

If your Amiga is PAL/50Hz, some monitors won't display 50Hz video although you usually get an "out of range" error message on screen.
Holding down the button during power on will force the output to 60Hz. If that works then select the refresh as 60Hz in the menu to set that permanently.
Also test with another TV / monitor if possible.

To test the SD card has been written correctly and the raspberry Pi zero is booting it properly, unplug the Pi from the board and apply power to it via the micro USB power input connector (socket closest to the corner) using a +5v micro USB power supply (phone charger etc).
You should get a menu screen with "CPLD Not Fitted" at the top which will indicate that it is booting correctly.

Make sure that all the pins are aligned on the Pi when plugging it back in to the board as being "off by one" will very likely damage the Pi when power is applied. 
Try another Pi if you have one.
There is also a serial debug output available if you have the necessary USB to serial hardware to hook that up.