hoglet67/RGBtoHDMI #339: Atari GTIA color Issue with scan lines enabled.

scorpio-ny, 2023-09-06:
Hi,

Using Lumacode with GTIAdigtizer, when the the scan line effect is one,  a screen in Atari 8 bit GTIA screen mode shows the wrong colors (a gradient is coarse and stepped). When the can line effect is off, it displays correctly

IanSB, 2023-09-11:
@scorpio-ny 
This is a bug which is due to variable intensity scanlines using half of the 8bpp colour palette so it only works if the palette generated by the computer uses 128 colours or less.
The Atari can generate 256 possible colours so uses all available palette entries with nothing left over for the scanlines.
To get variable intensity scanlines working will require changing lumacode to write to a 16bpp screen buffer.

In the meantime, try setting the FB Size setting in geometry to "Normal" or "Double Width" (i.e. disable double height)
This should force black only scanlines which don't require half of the palette.

BTW do you have a link to a program that demonstrates the problem as it is not apparent on everything.

r3mowilliams, 2023-09-12:
Ballblazer demonstrates this behavior.

What it looks like:
https://imagizer.imageshack.com/v2/1024x768q70/922/YZL8VU.png

What it should look like:
https://imagizer.imageshack.com/v2/1024x768q70/924/dbP93b.png

What it looks like with FB Size Normal/scanlines
https://imagizer.imageshack.com/v2/1024x768q70/923/DnE4u6.png

IanSB, 2023-09-19:
@scorpio-ny 
@r3mowilliams 

Please try this updated kernelrpi.img file:
[kernelrpi.zip](https://github.com/hoglet67/RGBtoHDMI/files/12654334/kernelrpi.zip)
Unzip and overwrite the file on the SD card and reboot

This currently works with black scanlines only as all 256 colours are used for the main palette leaving none for the variable intensity scanlines.

NTSC palette:
[image]

PAL palette:
[image]

I will investigate if it is possible to get multi level scanlines working using the 16bpp frame buffer

r3mowilliams, 2023-09-19:
Thank you, confirmed with that kernelrpi.img update.

[image]

IanSB, 2023-10-13:
@scorpio-ny 
@r3mowilliams 
There is a new beta60 here which incorporates the above fix:

https://github.com/IanSB/RGBtoHDMI/releases

Can you try a clean install to confirm everything is OK

r3mowilliams, 2023-10-19:
@IanSB 

Clean install of the Beta60 test release is displaying black scanlines when they are enabled as well as the correct GTIA shading.

Until there is a fix is it possible to create a separate profile that could be used to enable the variable intensity scanlines in the Beta60 (or subsequent) firmware?

r3mowilliams, 2023-10-19:
I've just loaded Alpha 61C and when scanlines are enabled they are variable intensity and I'm still seeing correct GTIA shading.

This is very welcome as black scanlines are something of a visual abomination these days :)