hoglet67/RGBtoHDMI #341: Why doesn't color autodetection not work for the Apple //e?

JPT77, 2023-10-03:
Hi, 

until now I had two Apple ][europlus, one modified to NTSC. Which makes it a a ][+
For both, color detection works flawless without any manual config. Textmode is B/W, gfx is colored. 

But now I got an Apple //e PAL. 
I only get Color when I set in Palette menu: 
- NTSC artifact (works for PAL apple?!)
- Artifact Color ON

but if I set this, the screen will always be interpreted as colored. which makes text screens difficult to read. 

what am I doing wrong?

is there any docs on this?
> The 4 lines of text in mixed modes will be clean when using the Apple IIc TTL interface
(The same signals are available internally on the Apple IIe)

thanks

Jan

IanSB, 2023-10-07:
@JPT77 

>But now I got an Apple //e PAL.
>I only get Color when I set in Palette menu:

If you are using the analog version of RGBtoHDMI plugged into the phono video output then artifact colour auto on/off is implemented by looking for the colour burst signal in the composite video as that is enabled in graphics mode and disabled in text mode. 

To get artifact colour on the PAL IIe you first have to disable the PAL transcoder which converts the NTSC artifact colour into something a PAL tv can display (this adds a 4.43Mhz colour subcarrier to the video which stops the pixel perfect sampling from working). However when this is done it also disables the colour burst entirely so there is no longer a signal to use to turn artifact colour on/off automatically and you have to turn it on/off manually using a long press of SW2 or the menu option.

There may be some way of modifying the PAL IIe so that auto detection works but my IIe is an NTSC version so I haven't been able to look at this.

>is there any docs on this?
>The 4 lines of text in mixed modes will be clean when using the Apple IIc TTL interface
>(The same signals are available internally on the Apple IIe)

You use a TTL version of RGBtoHDMI and connect to internal pins on the IIe as detailed in the cables section of the wiki:

Here's the entry from the cables section:

## Apple IIe digital interface internal pickup (1 + 1 bit TTL)  
This is preferred to connecting using the analog composite video interface as the 4 lines of text in mixed text and graphics mode will be displayed without artifacts. 
| 12 Bit Extender 16 way IDC | RGBtoHDMI 12 way IDC | Motherboard | Notes |
| --------- | ------------- | ----- | ----- |
| Pin 1  green0 | | | |
| Pin 2  red0 | | | |
| Pin 3  blue0 | | | |
| Pin 4  green1 | | | |
| Pin 5  red1 | Pin 1 red1 | | |
| Pin 6  blue1 | Pin 2 blue 1| | |
| Pin 7  GND | Pin 3  GND | UF5 166 pin 8 | 74166 GND |
| Pin 8  green2 | Pin 4  green2 | UE5 344-002x pin 2 | IOU GR - switches graphics on |
| Pin 9  red2 | Pin 5  red2 | | |
| Pin 10  blue2 | Pin 6  blue2 | | |
| Pin 11  RED3 | Pin 7  RED3 | | |
| Pin 12  HSYNC | Pin 8  HSYNC | UE5 344-002x pin 39 | IOU SYNC - Composite sync |
| Pin 13  GREEN3 | Pin 9  GREEN3 | UF5 166 pin 1 or pin 13 | 74166 SEROUT - TTL inverted video |
| Pin 14  VSYNC | Pin 10  VSYNC | | |
| Pin 15  BLUE3 | Pin 11  BLUE3 | | |
| Pin 16  +5V | Pin 12  +5V | UF5 166 pin 16 | 74166 +5V leave unconnected if powering Pi externally |

JPT77, 2023-10-13:
ehm. 
On my Apple //e F5 is a memory chip and E5 is a ROM labeled 342-0303-A...
so, are there differerent variants of the //e or did I misunderstand something?

JPT77, 2023-10-13:
This describes the different revisions of the //e
http://mirrors.apple2.org.za/apple.cabi.net/FAQs.and.INFO/A2.FAQs.MISC/A2.Prod.numbers.txt
I seem to have the //e enhanced, which is the second option in the list. 

here is a rev B schematics which shows your layout:
https://downloads.reactivemicro.com/Apple%20II%20Items/Hardware/IIe/Schematic/Apple%20IIe%20-%20Layout%20-%20Motherboard%20Rev.%20B.png
So "Rev B" seems to be an inaccurate naming...

I will try to find schematics and solve the problem myself. It's not your job. 
But I continue to document my progress here.

IanSB, 2023-10-13:
@JPT77 

I recall reading that the later revision European Apple IIe (Platinum board) had "International NTSC" rather than PAL which is a (non) standard that Apple came up with.
With this, Apple dropped the PAL transcoding and just used an NTSC style output but at 312 line 50Hz and with the slightly different PAL pixel clock frequency. This resulted in a hybrid video signal that had PAL video timings but NTSC colour burst with a non standard colour burst frequency that meant no monitor would produce a colour output except for Apple's special monitor that could handle the non-standard NTSC colour burst frequency.
(This is actually what the original PAL Apple II+ output without the colour card)

In theory this should just work with the analog board but I also recall that a US user also had issues with this later board due to different colour burst levels and had to adjust the colour burst detection level (DAC-B: G Lo in the sampling menu) to get colour detection to work.

Does your board actually produce a colour picture on a PAL composite monitor and is there a switch on the board to enable / disable PAL transcoding? If not then you may have international NTSC and perhaps adjusting the detect level above is all you need to do.
(If that's the case, post your saved profile and I will include it as an alternative IIe profile.)

Even if that works, it's probably still worth investigating the direct connection to get the clean 4 lines of text so if you do get it working, please post the connection details and I'll update the wiki with them.

JPT77, 2023-10-13:
I booted Alien Downpour and it is coloured on an old Sharp LCD TV and also with a cheap AV2HDMI converter which tells "PAL".
I don't have an SCART-Adapter just now, else I could test with a real CRT TV.

There is a switch that switches both monitors between bw and coloured. Cannot see any PAL/NTSC switch.

IanSB, 2023-10-13:
@JPT77 
>There is a switch that switches both monitors between bw and coloured. Cannot see any PAL/NTSC switch.

That would be the PAL transcoder on/off switch. In PAL mode you will get diagonal patterns on the video via RGBtoHDMI due to the PAL sub-carrier and switching that off will give a clean image but that appears to stop the auto detection from working.

When in the off mode, if adjusting the above mentioned DAC-B setting doesn't fix the auto detection then the only option is the direct digital connection so as mentioned you will need to determine the correct locations for your PCB.

Do you have a hires photo of your board?

JPT77, 2023-10-13:
Here is a photo online that looks very similar except the year on mine is 1984 not 82.
https://mirrors.apple2.org.za/Apple%20II%20Documentation%20Project/Computers/Apple%20II/Apple%20IIe/Photos/Apple%20IIe%20-%20Motherboard%20Front.jpg

There is a 74166 on C13.

IanSB, 2023-10-13:
@JPT77 
That's not the later CMOS/International NTSC  revision, just the PAL version which is again different to my original NTSC version.

Just pick up the signals from the 74166 (there is only one of those) and the IOU (which is labelled on the board and the chip is also marked 344-0022) from the pins indicated in the table.

JPT77, 2023-10-23:
Do you update the Cables section regarding the Chip positions?
it's UC13 and UC8.

IanSB, 2023-10-23:
@JPT77 
>Do you update the Cables section regarding the Chip positions? 
it's UC13 and UC8.

OK I will update that soon.

Did the direct connection work OK?
(I have not had any confirmation that this mod works on a PAL board)

JPT77, 2023-10-23:
I did not try it yet. Have to order some connectors first.

JPT77, 2023-11-09:
Works great. Pefect picture!