hoglet67/RGBtoHDMI #343: Cheaper or Alternative Chip for the XC9572XL-VQG44AWN

SensoryTrash, 2023-10-06:
Is there any cheaper or alternative chip for this build as keeping the cost down would be great given the amount of systems this can support.......

Thank you

IanSB, 2023-10-07:
@SensoryTrash 

The part number is XC9572XL-10VQG44C (not AWN)
There aren't any direct substitutes other than more expensive faster variants and in any case there aren't many alternative device types left with 5v tolerance these days. When first designed the CPLDs were about £3 each which was reasonable and they only went up in price due to the chip shortages over the past couple of years. They have started to come back down again but I don't know if they will ever get back to near their original price.

LinuxJedi, 2024-01-16:
@IanSB looks like the entire Xilinx CPLD line is about to hit EoL: https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/5783/XCN23009.pdf

We should maybe consider looking into something like the ATF15xx line.

IanSB, 2024-01-16:
@LinuxJedi 
>looks like the entire Xilinx CPLD line is about to hit EoL

Well that's annoying but not totally unexpected as most similar 5v tolerant devices have been discontinued.

>We should maybe consider looking into something like the ATF15xx line.

At least there are 5v versions of that but I wonder how long they will be around:

The ATF1504AS-10AU44 would seem to be a suitable physical substitute for maintaining the small zero sized footprint of the existing standard PCB but only has 64 macrocells and 32 I/Os vs 72 macrocells and 34 I/Os of the XC9572 so the functionality might have to be reduced and the loss of the 2 I/Os might be a show stopper although I have some ideas how to make it work with reduced I/O.
The alternative would be to move to one of the larger parts with more I/Os but they are quite a bit more expensive, unlikely to fit in the zero footprint and may be harder for DIY assembly.

LinuxJedi, 2024-01-16:
I figured size and I/Os may be an issue there unfortunately. I struggled to convert one of my other boards to one recently.

Only other ones I can think of are the Altera MAX II and V series, which will need buffers but have lots of space. Or Lattice, which I have never used and don't know if they are 5v tolerant. Beyond that I think we are looking at cheap FPGAs like the Efinix.

Either way I think level shifting may be the future.