hoglet67/RGBtoHDMI #345: RGB2HDMI with Amiga 500

shift838, 2023-10-13:
I have an Amiga 500 with standard Denise chip in it.  Would using the RGB2HDMI give me a better resolution like a super-denise or would it still be the same as a standard denise but just on a HDMI output?

IanSB, 2023-10-13:
@shift838 
It would be the same resolution as the standard Denise chip but the image would be noise free and very sharp on a 1080p monitor compared to a normal HDMI converter.

shift838, 2023-10-13:
that's what I was thinking but wanted to make sure.  I'll have to find me a Super-Denise to upgrade in my 500 to get some better resolution then.

venice1200, 2023-10-19:
Do you get higher resolution using a Super-Denise with a Stock-A500?

IanSB, 2023-10-22:
@venice1200 

I recently added support for some Super Denise modes (1280x256 & 640x480 productivity) but that requires a CPLD based RGBtoHDMI adapter, not the simple type adapter designed by c0pperdragon.

I don't know if a Super-Denise will work in a Stock-A500.

shift838, 2023-10-23:
I have read that it will and even found a video of one working.  I have a Super-Denise on order to find out.  It's coming from UK so probably be a couple of weeks before I see it.  Then I can tell you for sure if I get any extra modes or if the Amiga 500 will even boot up.

shift838, 2023-10-29:
Looks like there is a couple of other things that have to be done to a stock A500 to get the ECS modes.  Looks like the Agnus chip has to be replaced to, on top of the extra ECS modes the A500 will also allow up to 2mb of chip ram with the new Agnus chip.  There is also a trace that needs to be severed and a jumper connection that has to be moved.

You can review the video below; this is not my video but one I found, and I have emailed the author a couple of times.  Just with the new Super-Denise on the A500, the video does come up, but it has artifacts on the screen and only shows about 1/2 the desktop. 
 
[https://www.youtube.com/watch?v=mPwl_UfwkJM&lc=UgzPy3wJU7s0gevkDJJ4AaABAg.9wQnn3xMbU59wT7e8qZtPG](url)

venice1200, 2023-10-30:
Many Thanks for these information but too much to change from my point of view.

ryanm101, 2023-10-30:
To be honest you'd be better just getting an A600 or an A500+.
Stock 500 with ECS is the A500+ iirc that was the only diff between A500 and A500+.