hoglet67/RGBtoHDMI #346: Latest firmware does not work, previous firmware gives blue tint

salaxi54, 2023-10-13:
Hi, i've been trying Dec.22 firmware on my A500, and although everything else is fine, i get a bluish tint on startup (Workbench disk/hand screen), so no real white. This of course remains after booting into the OS.

IanSB, 2023-10-13:
@salaxi54 
What board are you using?

What image do you get when running the video test (F6) on the AmigaTestKit test program:
https://github.com/keirf/Amiga-Stuff/releases

salaxi54, 2023-10-14:
Thanks for replying! Here's some pictures to provide details.

[image]
That's the board

[image]
That's the video test screen

[image]
That's the blue-ish tint screen under HDMI

[image]
And that's how it shows using scart

salaxi54, 2023-10-14:
FYI, i've tested on various monitors, with results being the same. swapped HDMI cable, same thing.

IanSB, 2023-10-14:
@salaxi54 

The red gradient should look like the other two. There are four bits each for R, G & B (0-3) and the first four levels of red are OK which means bits 0 and 1 are working but the next four are just a repeat so it looks like bit 2 of the red channel is stuck low. This could either be a fault with the RGBtoHDMI board itself such as a short or open circuit or the connection between that board and the Pi zero or possibly a faulty Pi zero (badly soldered header or blown GPIO pin). The analog output is OK which means the Denise chip itself is working so the signals going to the RGBtoHDMI board input are correct.

Red bit 2 is on pin 22 of the Denise chip / 40 way DIL header so check for an open circuit around there (there won't be a short as that would affect the analog output as well)

Red bit 2 is then connected to U3 pin 4 on the RGBtoHDMI board and the output of that signal is on U3 pin 17 which is then connected to pin 32 of the 40 way Pi zero header so check for a short or open circuit around those areas.

See schematic:
https://github.com/hoglet67/RGBtoHDMI/blob/master/kicad_AmigaAdapter/V2/deniseadapter.pdf