hoglet67/RGBtoHDMI #347: Curious on picture differences

timmcross, 2023-10-16:
Hi there, love this product.  Been using it on a number of computers such as Atari ST, Coco, Apple IIgs, Apple II, Amiga.

I recently tried out the Apple II VGA card project that uses a RPI Pico (https://github.com/Retrotink/Apple-II-VGA) in the same Apple II I have an RGB2HDMI wired into.  The pictures they produce are different and I am kind of leaning towards the Pico-based product.  Any insight to why there is a difference?  Perhaps my settings are incorrect on the RGB2HDMI?

I still see NTSC artifacts near the bottom but the Exodus and Ultima lettering doesn't seem to show it on the VGA screenshot.  Perhaps the RGB2HDMI is a literal 1:1 representation where the Pico card is skipping some things?

RGB2HDMI (direct capture)
[image]

Apple II VGA Pico (sorry, had to use my phone)
[image]

IanSB, 2023-10-16:
@timmcross 

It's just different algorithms, some try to get as close to a real NTSC  TV and others produce a sharper image which is not necessarily the same as a real TV.

There is an NTSC Quality setting in the palette menu which can be varied from Soft / Normal / Sharp and that changes the output but introduces some black edges on the green:

[image]

Also if you have a Pi zero 2W fitted it uses another more accurate algorithm (to match real NTSC output):

Normal setting
[image]

Sharp setting:
[image]

I intend to do some work on the NTSC algorithms shortly so I will look at this as the Pico VGA is also open source so I might be able to port that algorithm to RGBtoHDMI as an option but that design snoops on the memory directly so it knows what video mode the Apple is in and the algorithm might depend on that. (RGBtoHDMI has to use a more generalised approach as it doesn't know what video mode is in use)

https://github.com/markadev/AppleII-VGA/tree/main/pico

timmcross, 2023-10-18:
Great!  Thanks for the info.  I will see if I can locate a Pico 2W too.

closed 2023-12-27 by timmcross