hoglet67/RGBtoHDMI #348: CPLD programing error TDO

AiGeekNet, 2023-10-21:
Hi, I have soldered my own issue 4 12-bit board, plugged the hat into a Pi Zero, I see the rainbow screen and CPLD Recovery Menu.

When I select CPLD_v79 and confirm, it return and nothing happened. I get the debug info below, any ideas what I may have done wrong? Thanks so much for any pointers!

Found CPLD: 6-12_BIT_BBC_CPLD_v79
Found CPLD: 3_BIT_BBC_CPLD_v68
Entering poll_keys_only, flags=00000000
Leaving poll_keys_only, result=0008
Entering poll_keys_only, flags=00000000
Leaving poll_keys_only, result=0002
Entering poll_keys_only, flags=00000000
Leaving poll_keys_only, result=0002
Read xsvf file /cpld_firmware/recovery/6-12_BIT_BBC_CPLD_v79.xsvf (length = 53373)
Programming....
CPLD programming: JTAG command = 0xfe (IDCODE)
 TDO Expected = 0xf9604093
 TDO Captured = 0xffffffff
 TDO Mask     = 0x0fffffff
ERROR:  TDO mismatch and exceeded max retries
ERROR at or near XSVF command #8.  See line #8 in the XSVF ASCII file.
Failed, error = 3
Entering poll_keys_only, flags=00000000

IanSB, 2023-10-21:
@AiGeekNet 
It's detecting the wrong type of board as you only get that menu with an issue 1-3 board and Issue 4 and later have 3 firmware options. It might be a soldering issue although if you got your CPLD from China it is very likely that it has been recycled and is already programmed with some other firmware which can stop initial reprogramming from working and can also cause the board to be mis-identified.

The fix for that is to cut the three solder bridges JP1, JP2 & JP4 then program the CPLD then remake the bridges.
This is mentioned in the CPLD menu with recent software but I don't think that message shows up for the old Issue 1-3 boards so I'll have to add it in a future release to cover mis-identification.

The programming menu on recent firmware should look like this:
[image]

AiGeekNet, 2023-10-21:
I have cut three jumers but get the same info and error.  Does it due to my CPLD IC is AWN ?

IanSB, 2023-10-21:
@AiGeekNet 
>I have cut three jumers but get the same info and error. Does it due to my CPLD IC is AWN ?

No, I have some with exactly the same batch codes and markings that work.

>Does it due to my board is 6bit ?

No, Issue 4 boards are 8/12 bit. You board is being mis-detected because a pin is held high instead of low.

First test that the jumpers are actually cut using a multimeter, especially JP1 as that isolates the signal that determines the board issue from a CPLD pin.

If you are sure that the jumpers are cut then check the soldering around Pin 9 of the CPLD, Pin 27 of the Pi header, Pin 7 of P5 and R7 (make sure it is the correct value). 
These are all connected together and are pulled low by R7 but the software is seeing this line pulled high on power up which is causing the incorrect identification.

See schematic:
https://raw.githack.com/hoglet67/RGBtoHDMI/master/Kicad_6Bit/v4/rgb-to-hdmi.pdf

IanSB, 2025-05-06:
@kennyevo 

For some reason LinuxJedi omitted the jumpers from his design so the only real option is to temporarily lift the legs of the CPLD instead.
The pins that have to be disconnected are CLKEN, SPDATA and SPCLK which are Pins 6, 7 and 44 of the CPLD.
See part of this thread which discusses it in relation to a fork of Linux Jedi's design by solarmon.

https://github.com/solarmon/RGBtoHDMI/issues/7#issuecomment-2096717958

Don't forget to resolder them afterwards.

kennyevo, 2025-05-08:
@IanSB 

Thank you for your help

Lifting the legs resulted in the new recovery menu screen where I can see the RGB and not just the BBC firmware. Unfortunately I cannot program it, when it wants me to confirm it just blinks and nothing happens, log says:

`Read xsvf file /cpld_firmware/recovery12/6-12_BIT_RGB_CPLD_v95.xsvf (length = 53373)
Programming....
CPLD programming: JTAG command = 0xfe (IDCODE)
Failed, error = 3`

I tried to erase the CPLD, got the following in the log:
`Re-creating /cpld_firmware/Delete_This_File_To_Erase_CPLD.txt
File /cpld_firmware/Delete_This_File_To_Erase_CPLD.txt re-created
Unknown CPLD: identifier = fff
Keycount = 0
CPLD  Design: 3-12_BIT_BBC
CPLD Version: f.f
Early erase of CPLD
Read xsvf file /cpld_firmware/recovery/blank/BLANK.xsvf (length = 157)
Programming....
CPLD programming: JTAG command = 0xfe (IDCODE)
Failed, error = 3
Early erase of CPLD failed`

I guess it's a defect in my good old chinese cpld

IanSB, 2025-05-08:
@kennyevo 
>I guess it's a defect in my good old chinese cpld

It could be faulty but if you are sure you have lifted those 3 pins and they aren't making any contact then it could also be soldering issues with the 4 CPLD pins that are connected to the Pi zero for programming.

Examine the soldering and also check the continuity between the following pins on the CPLD and Pi zero:

CPLD pin 9 and Pi zero pin 27
CPLD pin 10 and Pi zero pin 28
CPLD pin 11 and Pi zero pin 38
CPLD pin 24 and Pi zero pin 18

Also check that those CPLD pins are not shorted to adjacent pins which can sometimes not be visible if it is under the CPLD

You can also check that the lifted pins are no longer making contact 
Lifted CPLD pin 6 no contact with Pi zero pin 28
Lifted CPLD pin 7 no contact with Pi zero pin 27
Lifted CPLD pin 44 no contact with Pi zero pin 38

kennyevo, 2025-05-09:
@IanSB maybe you could help me with the next step, this is the image I see and auto calibration doesn’t work

[image]

IanSB, 2025-05-09:
@kennyevo 

You aren't getting any sync pulses
Have you selected the "Amiga" profile?

Check the connection between pin 32 of the Amiga's Denise chip and Pin 23 of the CPLD (Amiga sync to CPLD)
Then the connection between CPLD pin 20 and Pi zero pin 16 (CPLD sync out to Pi zero)

Check for open circuit or shorts to adjacent pins

kennyevo, 2025-05-09:
Well, would you think that the CPLD's 23pin was not soldered..... (that was the only leg that wasn't soldered properly)
Thank you again, it looks awesome now :D