hoglet67/RGBtoHDMI #349: can I use the RGBto HDMI with Korg PA800 music keyboard?

Persiandoc, 2023-10-31:
I have this old Korg Pa800 musical keyboard that  has a small 5-inch embedded display. The display is dim and old and parts can not be found on the market for it.
The device uses MC9328MX1 IC to control the LCD  display. Looking at the keyboard circuit it looks like it has a 8 line data , I assume 3-3-2 configuration for a 8 bit parallel RGB along DE, VS,HS and clock 

[image]

[image]

Is it possible to use this board to convert these signalls to HDMI

IanSB, 2023-10-31:
@Persiandoc 

It might work if those signals are as you assume but it would need a custom profile and possibly a custom palette.
The connections would be the 8 LD bits to the 8 video input bits on the header and the FLM and LP signals to Vsync and Hsync respectively.

There will be a new beta release due soon which has a create profile menu that makes it a lot easier to create custom profiles for applications like this (beta 61 or later):
https://github.com/IanSB/RGBtoHDMI/releases

If you have an oscilloscope maybe you could look at the FLM and LP signals to confirm they look like sync pulses. Also look at the clock pulse signal going to the LCD to measure its frequency as the create profile menu will need to know that.

Persiandoc, 2023-10-31:
Thank you Ian 
That is what I thought , I have ordered the board through https://retrohackshack.com/ and waiting to arrive to test it .
Per MC9328MX1 data sheet :
"The MC9328MX1 (i. MX1) processor features the advanced and power-efficient ARM920T™ core that operates at speeds up to 200 MHz."

I sold my Oscilloscope when I moved to California :(

I have tried this board from AliExpress  , and connected the LSBs to MSB but it did not work 
https://www.aliexpress.us/item/2255800855120005.html?spm=a2g0o.order_list.order_list_main.5.21ef1802p2fvD2&gatewayAdapt=glo2usa

I will keep you posted .
Thanks again.

IanSB, 2023-11-12:
@Persiandoc 
Beta 61 with the create profile menu now available:
https://github.com/IanSB/RGBtoHDMI/releases