hoglet67/RGBtoHDMI #350: 4K with Raspberry Pi 4

tyristori, 2023-11-03:
Does RGBtoHDMI support 4k/50hz/60hz when used with Raspberry Pi 4?

IanSB, 2023-11-05:
It didn't but it will on the next beta release due soon:
[image]

stanstrup, 2024-01-01:
Would someone be kind enough to confirm what is needed for 4K with a C64?

I found this:
* https://www.tindie.com/products/c0pperdragon/vic-ii-dizer/
* https://www.tindie.com/products/c0pperdragon/rgbtohdmi-mono-lumacode/

At the tindie page it says that the only extras that are needed are: "Micro SDcard, Mini HDMI cable, USB power supply".
But the hardware section of this repo lists additional things. Am I right in assuming that the things listed there are build into the rgbtohdmi-mono-lumacode?

IanSB, 2024-01-02:
@stanstrup 
>At the tindie page it says that the only extras that are needed are: "Micro SDcard, Mini HDMI cable, USB power supply".

Yes, that is all you need plus a Raspberry Pi of some type. Normally that is a Pi zero but a Pi 4 is required for 4K support @50/60Hz and the cheapest one of those with 1GB memory will work OK with the mono/lumacode board. 

The mono/lumacode board has part of the analog board already built into the main PCB and that is just enough to support mono analog sources like Apple II and lumacode like c0pperdragon's VICIIDizer so no additional boards are required.

If you use a Pi zero when connected to a 4K monitor it will output 1080p@50/60Hz by default but you can manually select 4K@25/30Hz which is the maximum that the zero supports. This will give 4K resolution but the motion will be limited to 25 or 30Hz depending if the source computer is 50 or 60Hz

You need the latest beta software from here for 4K support:
https://github.com/IanSB/RGBtoHDMI/releases

Note that using some other RGBtoHDMI board variants with Pi models other than the zero / zero2W requires a 40 way GPIO extender to allow clearance for connectors but the above board should plug into the Pi4 without that.

stanstrup, 2024-01-02:
Thank you! That is super useful information! If you are not well-versed in the Pi ecosystem this can all be a bit confusing.
I guess now I have to see what a c64 looks like in digital 4K!

IanSB, 2024-01-02:
@stanstrup 
I just remembered there are few differences with the Pi 4:
The Pi 4 requires a USB-C power supply (the zero uses micro USB power)
The Pi 4 requires a micro HDMI lead, not a mini HDMI lead (Use the HDMI socket closest to the power socket)

tyristori, 2024-01-02:
I've been using c0pperdragon's video enhancement mod with RGBtoHDMI and Pi4. 4K/50hz works great and looks stunning on my LG OLED tv.

I hacked together a riser by soldering two female 2x20 pin headers together so I could fit the RGBtoHDMI board to Pi4. Other option would be to remove network and usb connectors from Pi4.

I added TRRS connector to Analog board pcb so I could use a premade TRRS to TRRS cable.

[image]
[image]
[image]
[image]
[image]

stanstrup, 2024-01-03:
@tyristori Looks very cool! If I understood @IanSB correctly I wouldn't need the extender you made for the particular board I am targeting.

Your comment about TRRS cables made me think about the audio. How is that handled? I cannot see on c0pperdragon's "RGBtoHDMI Mono & LumaCode" board if it has an audio input. So how should I get the audio into the HDMI signal? I have a small s-video/RCA board for the C64 so I have the analogue audio signal but what do I do with it? I can plug it into the RGBtoHDMI board? The Pi? Or would I need to merge them using an HDMI audio embedder (seems at least ~50€) myself?
#340, #254, #225 and #177 would suggest to me that I indeed would need an audio embedder. Also am I right in assuming most TVs won't allow a separate audio source when using HDMI?

@IanSB Also thanks for the additional hints. I think I have those parts lying around.

IanSB, 2024-01-04:
@stanstrup 
> I wouldn't need the extender you made for the particular board I am targeting.

That is probably the case but with one caveat (see below)

>Your comment about TRRS cables made me think about the audio. How is that handled?

Audio isn't handled at the moment so you either have to use separate speakers (like a soundbar), an analog audio override connection if your TV / monitor supports that or a HDMI audio embedder.

I am looking at adding audio support in future but it is difficult because there is not much hardware documentation for bare metal programming on the Pi.

Full 4K 50/60Hz support at the moment is very new and has had little testing so it's a good thing you mentioned that as I hadn't actually got around to testing my Pi 4 with an audio embedder (which I have now done)

First, before going down the 4K path you might want to check how your TV/monitor scales 1080p content.
Most TVs use interpolation so 1080p will have soft edges on a 4K monitor and this is most noticeable when using an older PC with a 1080p desktop as window edges etc will be soft compared to a normal 1080p monitor.
However some newer 4K monitors are supposed to have integer scaling for 1080p to 4K by just replicating each 1080p pixel in a group of 2x2 pixels on the 4K monitor which would make a 1080p PC desktop look as sharp as a normal 1080p monitor. (This is the same thing that RGBtoHDMI does for vintage computers).

I don't know how common this is but there is some more info here
https://www.reddit.com/r/integer_scaling/comments/ohpkno/eve_spectrum_worlds_first_computer_monitors_with/
https://shmups.system11.org/viewtopic.php?t=67116
Such a feature might have to be enabled in a menu
If your TV/Monitor does that then there is no point going to 4K as 1080p will look identical and you can stick with a Pi zero.

>Also am I right in assuming most TVs won't allow a separate audio source when using HDMI?

You will need to check your TV's manual.
Some TV's and many monitors have a 3.5mm audio jack input which will override the HDMI audio on one input when connected and that is designed for use with older PCs that had an analog audio output separate from the monitor output but maybe that is less common these days.

I have this audio embedder which is rated for 4K@60Hz:
https://www.aliexpress.com/item/1005005604559797.html

I have tried this and it works but there are some issues which will need investigation:
When the embedder is connected to one of my newer Pi 4 models it works OK but connected to the older one it gets stuck in a reboot loop although I found a way around that by temporarily removing the embedder and forcing the Pi's resolution to Full 4K(Pi4 Only) rather than Auto in the main menu. 

Also it is very sensitive to cable quality and length so use HDMI 2.0 rated short cables (I used 50cm from Pi4 to embedder and 1M from embedder to TV)

I also added a small stick on heatsink to the Pi 4's CPU and that might mean you need a GPIO extender depending on the height of the heatsink
Both of these are available from Pi sellers and places like ebay and amazon e.g:
https://thepihut.com/products/40-pin-extra-tall-header-push-fit-version-single-shroud
https://thepihut.com/products/40-pin-extra-tall-header-push-fit-version-no-shroud
https://thepihut.com/products/xl-raspberry-pi-4-heatsink

stanstrup, 2024-01-17:
So I got everything together apart for the audio embedder.
I am getting some garbled output though... Any ideas?
I left my lumafix in there (below the vic-ii-dizer). Could that cause this issue?

[image]

IanSB, 2024-01-17:
@stanstrup 

If it's affecting the S-Video output as well then it is likely that some VICII signal or signals are not connecting properly or being held at the wrong voltage by the lumacode board and maybe putting a different 40 pin socket between the lumcode board and the c64 socket would help to fix that but it's best if you post about issues with the lumacode board itself to c0pperdragon's github:

https://github.com/c0pperdragon/LumaCode/issues

I can really only help with the RGBtoHDMI board although there does seem to be a problem with that as well as your Pi seems to be outputting 4K@30Hz rather than 4K@50hz but it's probably better to look at that once your lumacode board is working.
(Although maybe that would be expected if you are doing initial testing with a zero rather than a Pi4)

BTW I did find some issues with the audio embedder at 4K and there will be a Beta 63 release soon that fixes those.

stanstrup, 2024-01-18:
Thanks. I have ordered some sockets and hope to be able to fix this as soon as I get them.

stanstrup, 2024-03-08:
Turns out in all these trials my vic chip died :( 
After replacement all is working well and it looks fantastic.
I don't dare try again to add the lumafix together with this.