hoglet67/RGBtoHDMI #353: Having issues with configuring the RGBToHDMI for the denise adaptor board

ADoozer, 2023-11-21:
Hi all.

posted this on c0pperdragon issues as well so apologies for duplicate post.
(https://github.com/c0pperdragon/Amiga-Digital-Video/issues/70)

I am using the c0pperdragon amiga 500 direct denise adapter (V2) (no CPLD) but every time I boot the pi I see the message "Release buttons for CPLD recovery menu"

I have tested voltages, and the latch chips, I'm seeing the 15khz and 7MHz on pins 11 and 16 (cant remember which way around it is but It looks ok?!), and my monitor reports 50Hz (UK Amiga)

Watching Jan Beta, CRG, brostenen etc on utube, they just download the zip, send it to the SD and it just works.

Pretty sure its just a setup issue?! but no idea where to start looking.

Anywho, Any help greatly appreciated.

Regards Dave.

IanSB, 2023-11-21:
@ADoozer 
>I am using the c0pperdragon amiga 500 direct denise adapter (V2) (no CPLD) but every time I boot the pi I see the message "Release buttons for CPLD recovery menu"

That board should work without any additional configuration. (The linuxjedi internal CPLD variant does require some manual configuration) 

To get the "Release buttons for CPLD recovery menu" message requires all three buttons to be pressed during powerup and two of those GPIO pins are not even connected to anything on the c0pperdragon board (pins 35 & 37 on the Pi header) so something is wrong with the hardware.
e.g. maybe the zero was plugged in backwards or "off by one" at some time which damaged the GPIOs so it is seeing those inputs as low.

If you have a TTL to USB serial adapter you could connect that to the serial debug output to see what it is detecting and post the resulting output.

Pin 8 on the zero header is TxData and Pin 6 is GND
115200 baud 8 data no parity

ADoozer, 2023-11-21:
Thanks Ian.

Looks like a self inflicted wound.

I was testing on my 4b while i wait for my zero to turn up (hopefully today/tomorrow) and made the poor assumption that all the GPIO was equivalent. 

However I now know (after reading the docs) that the higher GPIO pins are internally pulled low at initialisation resulting in "all buttons pressed"

I cant find anything that explicitly says so, but I'm guessing the zero has internal pull ups on the higher GPIO pins at initialisation

Thanks again Ian

Regards Dave

IanSB, 2023-11-22:
@ADoozer 

The GPIOs do default as you indicate and I think that is the same for all Pi models but the software does reprogram them to pullups before testing the GPIOs. However those pullups are very weak so maybe they haven't had a chance to fully pullup before the software reads the GPIOs and the time taken for that is probably different between the Pi zero and the Pi 4 as the latter is faster. The standard CPLD board uses strong 1K pullup resistors on the switch inputs so doesn't rely on the internal weak ones which means any difference between Pi models wouldn't be noticeable with that board.

ADoozer, 2023-11-26:
Thanks Ian.

Just confirming the zero worked as advertised with no changes required

Regards Dave

hoglet67, 2023-11-26:
Ian,

The mechanism for controlling GPIO pullups/pulldowns is different on the Pi 4 compared to the other models. So it's possible this is a software bug. Take a look at the code in PiTubeDirect here:
https://github.com/hoglet67/PiTubeDirect/blob/master/src/rpi-gpio.c#L118
```
void RPI_SetGpioPull(rpi_gpio_pin_t gpio, rpi_gpio_pull pull)
{
#if defined(RPI4)
  rpi_reg_rw_t* pull_reg = &RPI_GpioBase->GPPULL[gpio / 16];

  rpi_reg_rw_t pull_copy = *pull_reg;
  pull_copy &= (uint32_t)~(0x3 << ((gpio % 16) * 2));
  pull_copy |= (pull << ((gpio % 16) * 2));
  *pull_reg = pull_copy;
#else
  RPI_GpioBase->GPPUD = pull;
  RPI_WaitMicroSeconds(2); // wait of 150 cycles needed see datasheet

  RPI_GpioBase->GPPUDCLK0 = 1<<gpio;
  RPI_WaitMicroSeconds(2); // wait of 150 cycles needed see datasheet

  RPI_GpioBase->GPPUDCLK0 = 0;
#endif
}
```
You can also configure pullups in the config.txt file using the following syntax:
```
# Configure pullups on D6,D4,D2,D0,nTUBE
gpio=24,22,10,8,17=pu

# Configure pulldowns on D7,D5,D3,D1
gpio=25,23,11,9=pd
```

IanSB, 2023-11-26:
@hoglet67 
>The mechanism for controlling GPIO pullups/pulldowns is different on the Pi 4 compared to the other models.

Thanks, you are correct that the pi4 gpio pullup code is missing so I have merged the PiTubeDirect code to fix that.

See here:
https://github.com/hoglet67/RGBtoHDMI/commit/df4ec94542666395423abd11b3e3bdcc823acc18