hoglet67/RGBtoHDMI #355: Flashing screen using NTSC C64 Lumacode

jc13git, 2023-12-01:
Hello - I have just installed the VIC-II Lumacode board in an NTSC C64 
[image]
[image]
[image]

and am having trouble with the screen flashing once a second or so (unless the menu is on-screen).  This seems to be similar to an Amiga [issue](https://github.com/hoglet67/RGBtoHDMI/issues/206) from a couple of years ago.  I'm a new user of the RGBtoHDMI, so it's likely that I'm doing something wrong, but when I select C64 Lumacode and the 60Hz sub profile it looks great - until I exit the RGBtoHDMI menu.  That's when the flashing starts.  Also, I'm not sure if the RGBtoHDMI should be autodetecting the profile, but it doesn't do that at reboot (the pictures show auto_switch is off - this was later on in the troubleshooting).  It always defaults to the 50Hz sub-profile.  Once other piece of information is that the Lumacode board has a "special" firmware on it due to my VIC chip being the earlier 6567R56A variant.  Maybe there needs to be a special profile for that chip?  Pictures are attached showing a little more information.  Thanks for any suggestions and if any other info would help, let me know and I'll gather it.

c0pperdragon, 2023-12-01:
Hm, maybe I have mixed up the numbers, and it is in fact 1024x262 for the "special" NTSC and 1040x263 for "usual" NTSC.

IanSB, 2023-12-01:
@jc13git 

First wipe your SD card and install the latest beta from here:
https://github.com/IanSB/RGBtoHDMI/releases
Then download this file:
[C64LC_R56A_60Hz.txt](https://github.com/hoglet67/RGBtoHDMI/files/13530881/C64LC_R56A_60Hz.txt)

and put it in \Profiles\6-12_BIT_YUV_Analog\Commodore_\Commodore_64_Lumacode_ on the SD card and reboot the Pi

You will need to centre the image by adjusting the H and V offsets in the geometry menu, then save the configuration and post the completed profile for inclusion in the next release.
Saved profiles will be in /Saved_Profiles on the SD card.

jc13git, 2023-12-01:
@IanSB and @c0pperdragon 

Initial testing looks good.  I'll get the lumacode board wired in more permanently and do some more testing, but it seems to work.  It also auto-detected the sub profile too.  Thanks to both of you for your help and building these awesome pieces of hardware and software!

tyristori, 2023-12-02:
Sorry for the non topic question but how do I edit the default Commodore 64 palette in RGBtoHDMI?

I can't find any palette related files in the software archive.

Default C64 palette is a bit too saturated for my taste and I would like to edit it.

IanSB, 2023-12-02:
@tyristori 

The palette files are generated programmatically during first boot up so will appear in the /Palettes folder on the SD card (The c64 uses the colodore algorithm). You can manually edit those files after that or create new ones and put them in that folder and they will also appear for selection after subsequent rebooting.

You can also change the overall palette by varying the settings like saturation in the palette menu

I haven't actually got around to documenting creating your own palette files but it is pretty straightforward:
The palette table is 256 32 bit words plus an additional 32 bit word on the end containing the number of palette entries that are active (this will vary depending on the computer).

Each 32 bit palette value is 4 bytes (0-255) in the following order:
red, green, blue, mono (the luminance value used in mono mode)
So for the C64 the first 16 words would be the colours and the 257th word would be 0x00000010 (16)
Generally you should repeat the palette data for the rest of the 256 entries.

There is a slight problem with the C64 lumacode as the order of the colours so it doesn't match the standard C64 colour order so you have to take that into account when creating the table.
See the table at the bottom of this page:
https://github.com/c0pperdragon/LumaCode/wiki/VIC-II-dizer-(for-the-C64-computer)