hoglet67/RGBtoHDMI #357: Flashing screen/unkown mode on EC1834

christianhorn, 2023-12-20:
I try to use MCE2HDMI on a Robotron EC1834 (an east German IBM clone), with K7033.10 COL graphics card. The few details I know:

- The card does up to 640×480, is CGA- and EGA-compatible
- The card can do 32kHz vertical frequency, 29,58kHz horizontal frequency, 58.4-62.4Hz picture frequency

I have gone through the quickstart docs, and tried out the "IBM compatible" profiles. When leaving the menus, the HDMI output is flashing between something looking like mangled console output, and a black screen. The CLK led is also flashing with the same frequency. Starting with the base profiles did not bring me anywhere either. Any hints? The graphics card might also be defective, I got the system with a monitor - but the monitor is not showing anything either. 
[image]
[image]
[image]
[image]

IanSB, 2023-12-23:
@christianhorn 

It looks like Vsync is being detected correctly as the status is showing the frame rate as 59.97Hz but Hsync is not correct as it is showing only 161 lines per frame rather than 262 for CGA or higher for EGA so that is probably why the image is not locked horizontally.
Do you have any way to look at the pulses being generated on the video connector to confirm that sync pulses are on the right pins e.g. using an oscilloscope or USB logic analyser?

christianhorn, 2023-12-24:
Thanks a bunch for these details!
I have no oscilloscope or logic analyser at hand. 
Is MCE2HDMI supposed to deal with this? If so, is it possible to hint in the software on the mode which (supposedly) is fed into the device?
"supposedly" as the hardware might be defective. I'm just within reach of the hardware in the next ~7 days, then it will probably be months until I next time get access.
The overall options I keep in mind (apart of potential MCE2HDMI software tuning): 

- asking someone with a osci to come around and help
- getting a USB logic analyser myself (they seem quite affordable.. recommendations appreciated)
- when I find someone with a fitting monitor, or a comparable graphics card (it's not normal ISA connector, I guess this pins it down to a graphics card from another EC1834), we could replace-and-see-if-it-works

closed 2025-08-01 by christianhorn