hoglet67/RGBtoHDMI #358: TI99 4/A TMS-RGB Palette..

dabonetn, 2023-12-23:
I'm trying to get a NTSC TI99/4a working with the RGB2HDMI using a TMS-RGB.
I've got the latest analog board from Retrohack shack.

I'm getting a nice stable picture using a modified Camputers Lynx profile, But the colors are off.

sampling=5,5,5,5,5,5,5,0,1,0,5,0,0,0,0,0,0,0,0,0,100,256,100,256,100,256,256,256
geometry=60,25,256,252,280,286,3,2,3,0,6000000,384,5000,312,4,0,0

Screen Capture.
[image]

Emulator output showing proper colors.
[image]

Suggestions on where to get started?

IanSB, 2024-01-02:
@dabonetn 

Getting the TI99/4a fully working with an analog RGBtoHDMI is quite difficult because the video voltage levels are a bit too close together to separate reliably. I have got it fully working both with a direct connection to the YUV output and using the TMS-RGB but the levels need adjusting as it warms up so it is not really a good solution.

However a lumacode board for the TI99 and other systems using the same video chip like the Colecovision has been developed and I now have a prototype for testing so that should be available in the future.

dabonetn, 2024-01-02:
I just saw about the TMS board yesterday on his github. 
Can't wait.

IanSB, 2024-01-22:
@dabonetn 
FYI:
https://github.com/c0pperdragon/LumaCode/issues/13#issuecomment-1902838176

dabonetn, 2024-02-09:
Lumacode bought and installed... Works great. This can be closed.

closed 2024-02-09 by dabonetn