hoglet67/RGBtoHDMI #359: Is there a way to verify that I soldered the board correctly?

dfroholt, 2024-01-13:
My first attempt at SMD soldering went way better than I expected. At least I think it did. Is there a way to verify that I soldered everything together correctly without an actual computer to test it with?

I powered the unit on and programmed the CPLD without any issues but when I hooked it up to my old Commodore PC10-2 I only get a grey screen and no sync. It's been over 25 years since this computer was powered on. Back then I remember thinking that the monitor stopped working because I did not get an output from the computer.

Now I am thinking that there's an issue with the computer itself. Perhaps the ATI CGA card. OR I messed up the soldering. I would like to know for certain but I don't have any other PCs to test it with. I got a some stock C64s and a couple of Sharp MZ-700 but I don't really know how to use the RGBtoHDMI board with those. The next step would probably be to research this.

I stuck my ocilloscope into the connector on the video card and I can't really see any activity. Pin 3, 4, 5 and 9 seems to be stuck high at around 3.7v. I don't really know what to expect here though but it seems a bit weird to me.

IanSB, 2024-01-13:
@dfroholt 
>I got a some stock C64s and a couple of Sharp MZ-700 but I don't really know how to use the RGBtoHDMI board with those.

The C64 will not work with a digital RGBtoHDMI (although it will work with an analog RGBtoHDMI and one of c0pperdragon's lumacode boards) but the Sharp MZ700 will work and there is already a profile available in the software.

The pinout of the sharp RGB connector is shown here:
https://original.sharpmz.org/mz-700/connect.htm

The connections for the MZ700 are shown here:
https://github.com/hoglet67/RGBtoHDMI/wiki/Cables#sharp-mz700-3-bit-rgb-ttl

Note at the moment this profile is set up to use the composite sync output but looking at the above Sharp pinout website that wouldn't work with the MZ800 so I may update that in future to work with separate H and V syncs.

dfroholt, 2024-01-13:
Thanks! I'll try to make a cable for the Sharp. Hopefully it will work!

closed 2024-01-17 by dfroholt