hoglet67/RGBtoHDMI #362: Heath/Zenith Z-110 - unable to sync

hmb001, 2024-02-06:
For reference, I am trying to connect the Zenith Z-110:
https://z100lifeline.swvagts.com/articles/Z100_Monitors.pdf

I have no problem connecting the RGB output via an MCE2VGA adapter to a VGA monitor - no special settings required, it simply just works.

Not so with the RGBtoHDMI. No matter what settings it try on the RGBtoHDMI adapter, it fails to sync to the incoming signal. The screen just flickers with parts of the image visible but jumping all over the place.

The Z-110's video output itself is very configurable:
- horizontal sync can be set to negative (default) or positive polarity
- can be set to vertical sync (default) or composite sync
- vertical sync can be set to negative (default) or positive polarity
The horizontal scan frequency is 15KHz

I have tried all combinations of settings for the Z-110 video output signal, and for each combination I have tried every combination of sync signal type on the RGBtoHDMI adapter. Unless I am missing something, it would seem this is an application RBGtoHDMI adapter is unable to cope with, whereas the MCE2VGA adapter was able to work out-of-the-box.

IanSB, 2024-02-08:
@hmb001 
>the Z100 horizontal scan frequency is 15.75 KHz
>the video output is not exact IBMPC CGA standard - it is 640x225 (vs 640x200 for IBM PC CGA)

RGBtoHDMI is primarily designed to upscale ~15Khz type sources and is fully configurable for any unusual resoloutions so should be able to work with such a source.

>Am I correct in assuming that since the RGBtoHDMI adapter is not able to function as a scan doubler, it is necessary to use an >HDMI monitor capable of handling a 15.75 KHz horizontal frequency if the RGB video output of the vintage computer is 15.75 KHz

No, that is not correct, RGBtoHDMI is a scaler that will scale the video to the native resolution and refresh rate of the LCD display you are using. e.g. If connected to a 1080p monitor it will scale a 15Khz source to a 4:3 image pillar boxed in 1920x1080 at 50Hz or 60Hz as appropriate usually with integer scaling for a very sharp image.
It should work with any HDMI or DVI monitor and doesn't require the monitor to support 15Khz.

As to why your source doesn't work, do you get a message at the top of the screen when the menu is enabled that says "No Sync Detected?"

Possible causes for that could be:
1. Wrong sync type configured (e.g. source is composite but the profile is configured for separate H/V)
For such a source I would start with the CGA profile but you can change the sync setting in the geometry meny to cycle between the various separate and composite sync options if you haven't already tried that.
 
2. Sync voltages are not at TTL level (e.g. 1v p-p instead)

3. Faulty RGBtoHDMI
Have you been able to use the board with any other CGA source?

hmb001, 2024-02-08:
There is no error message indicating the device is unable to detect sync. The only error message is that the device is unable to establish genlock.

After a lot of fiddling, what I see are two unstable images of the screen display, one above the other. The lines are badly distorted and compressed together (ie completely illegible) and fill the entire width of the screen.

1. The source is definitely separate H and V sync (both positive) with the standard pin assignments on the connector. On the adapter, I have selected the CGA profile. In the geometry screen, I have manually selected all combinations of H and V sync (in addition to H and V being both positive per the design of the source) with no improvement.

2. All signals generated by the source are driven directly from standard TTL logic.

3. I have not tried the RGBtoHDMI board with any other source... I figured it wasn't designed to work with standard IBM-PC CGA output without use of a monitor capable of 15KHz horizontal sync. However, now that you say it should work I will test with the output of a standard PC CGA board.

IanSB, 2024-02-08:
@hmb001 
>After a lot of fiddling, what I see on the monitor are two unstable images of the same screen display, one above the other.

Can you post the source summary page in the info menu
Either a photo or a screencap (press up and down buttons simultaneously in the menu, file in /Captures on SD card)

Also try the create profile menu in the latest beta which attempts to create a profile automatically by analysing the timing. 
(This will need valid syncs before you can enter the menu)

Download here:
https://github.com/IanSB/RGBtoHDMI/releases

Select the CGA profile as your base profile then enter the create profile menu and follow the help menus (assuming it can detect syncs).

IanSB, 2024-02-08:
@hmb001 

Looking at the schematic of the Z100 video board, the pixel clock is 14112000Hz instead of 14318181Hz so you will need to change the clock frequency setting in the Create Profile menu to match that.

hmb001, 2024-02-08:
Re changing the pixel clock - I will do that, and let you know the outcome.

I have a few other options to dig into this problem further knowing this setup *should* work, like putting a scope on the H and V sync signals to see what is going on there.

hmb001, 2024-02-09:
If I have this right, to create a custom profile:
- Main Menu -> Profile -> Select "Custom"
- Geometry Menu -> select "Clock Frequency" and "Sync Type"

I see no "Create Profile" menu anywhere. For that matter, navigating to a "Create Profile" menu item is not shown anywhere in the Wiki.

Attached are:
- photo of my Geometry menu
- photo of my Source Summary menu
- logic analyzer sample showing V and H sync pulses at input to the RGBtoHDMI board, with expected frequencies.

[image]
[image]
[image]

hmb001, 2024-02-10:
Full disclosure - I am using the version of this adapter sold by TexElec.

I hooked up the adapter to a "real" CGA video output card... and I am happy to say it works just fine. This adapter has none of the residual "shimmering" around pixels in a character that I see with the several different FPGA-based RGB-to-VGA adapters I have used.

Having ruled out a hardware problem with my RGB-to-HDMI adapter, I still do not have an answer as to why the CGA-like output of the Zenith Z-100 cannot be displayed. As you pointed out, testing so far seems to indicate the adapter is unable to lock on to the H sync signal. Perhaps there is some subtle issue with the timing of the H sync signal, and the Pi is unable to correctly identify the H sync signal because it is too long or short?

At this point I'm leaning towards dropping this project - perhaps with current programming I happen to have a vintage computer with a CGA-like signal this adapter is incompatible with. As I mentioned I do have a couple of different RGB-to-VGA adapters (both of which do work just fine with the Z100) and I can just continue to us them instead.

IanSB, 2024-02-11:
@hmb001 
>Perhaps with the current Pi software I have a vintage computer with a CGA-like signal this adapter is incompatible with.

This is extremely unlikely as the software works with a wide range of refresh rates and sync pulse sizes.
Correct me if I'm wrong as it is difficult to judge from you photo but the time between hsync pulses looks to be around 64uS and the sync pulse width looks to be around 4.7uS which are pretty much standard. (The reading of 103190ns is actually a timeout value meaning no h sync pulse edges have been seen at all.)

>I am using the version of this adapter sold by TexElec. 
>I hooked up the adapter to a "real" CGA video output card... and I am happy to say it works just fine.
>Having ruled out a hardware problem with my RGB-to-HDMI adapter

This is almost certainly a hardware issue as there is no reason why sync signals with the timing you posted shouldn't work and produce equally high quality output with an appropriate custom profile. Perhaps the termination on their hsync input is too low resistance for the Z100 to drive or the hsync pin on your cable isn't mating properly with the socket on the converter but you will need to get support from your supplier for that.

IanSB, 2024-02-11:
@hmb001 

One final suggestion:
It appears that you can change jumpers on the Z100 to output composite sync on the Vsync output. As that input appears to work judging by 60Hz being detected you could try changing the jumpers then alter the CGA profile to look for composite sync on the vsync input as follows:
Change "Sync Type" in the geometry menu to "Composite" (for negative going composite sync) or "Inverted" (for positive going composite sync)
Change "Sync on G/V" in the sampling menu to "On" which will select composite sync on the Vsync input instead of the Hsync input.

hmb001, 2024-02-11:
The logic analyzer trace I provided was directly at the termination resistor on the adapter board, which would rule out a connector problem. Also, the logic analyzer was able to interpret that signal as a logic "1" although that does not necessarily mean the LS245 input buffer sees the signal as a logic high.

I have not been able to find any PDF schematic for your 6 bit buffer board design (only the Kicad files, which I don't have the app for), but based on the BOM it would seem the buffer board on my TexElec adapter is identical. I looks like all the signals coming into the RGB connector are terminated by a 4.7K load resistor to ground, and to the inputs of an LS245 buffer. I have not traced the signals past the LS245, but given the adapter works fine when connected to another system it is unlikely there is a circuit fault there.

Over the past few days I did try setting the Z100 video output to both negative and positive composite sync, with no success. What I did not do is set the adapter to "Sync on G/V" - I had assumed setting the sync to "composite" would have sufficed for the adapter to figure it out, but apparently not. I will need to give that a try. Also, thank you for the pointer on what the "sync type" setting does - that is not described in the Wiki, and who would have guessed that "inverted" meant positive going composite sync.

IanSB, 2024-02-11:
@hmb001 

As you have been experimenting for a while make sure you delete the /Saved_Profiles folder on the SD card or at least use the "restore default configuration" option in case you inadvertently saved messed up settings to the CGA or other profiles which will happen if you used an auto calibration followed by the press menu to save option.

hmb001, 2024-02-17:
I am happy to report I have resolved the problem.

For some inexplicable reason, the H and V signals going to the Z100 external monitor connection are each driven by a TTL open collector driver with a 1K pullup resistor to 5V. This stupid design provides nowhere near enough drive current to get a clean video signal to the RGBtoHDMI adapter.
Fortunately, the Z100 video board provides an optional output connector for just the H and V signals, with each signal driven independently of the above connector driven by a 74LS244 driver. I re-routed the H and V signal outputs to this connector.... and the RGBtoHDMI adapter immediately synced.

Thank you for your encouragement and advice getting this working.

IanSB, 2024-02-18:
@hmb001 
>This design provides nowhere near enough drive current to get a clean video signal output

I had noticed that but thought it should be enough for the required voltage swing as it is a 1K pullup on the Z100 and a 4.7K pulldown on RGBtoHDMI which should overall give about a 4V logic high. However if the Hsync open collector output is faulty and not pulling down to ground hard enough then that might explain the issue as well because the Vsync output appears to work. Perhaps Hsync is only pulling down enough for CMOS levels so that would explain why your logic analyser and VGA converter still work.

>RGBtoHDMI adapter immediately synced using the "Tandy 1000 225 line" profile setting
>There is just one one minor issue I still have to figure out, which is that the horizontal width of the display is short by one or two pixels.

Syncing is only the first step. You will get an image but it might have columns of noise on some characters across the screen if the pixel sampling is set incorrectly and other parameters like lines per frame are also likely set incorrectly.

This is a non standard source so you need to setup a custom profile using the latest software version with the "Create Custom Profile" menu and also follow the newly updated (a few days ago) Tutorial on creating a new profile to ensure you get a perfect noise free accurately sampled image.

First download the latest stable release from a couple of days ago:
https://github.com/hoglet67/RGBtoHDMI/releases

Then re-read the newly updated tutorial on creating a custom profile:
https://github.com/IanSB/RGBtoHDMI/wiki/Tutorial-on-Adding-a-New-Profile
(The rest of the wiki has also been updated a few days ago to document recent changes)

I think the Clock Frequency needs to be set to 14112000Hz instead of 14318181Hz as that is the crystal on the schematics and you also need to fill the screen with 'M' or similar as described in the tutorial and if necessary adjust line length to ensure that all characters across the screen are being sampled correctly.

If you get a satisfactory custom profile, please post it here and I will include it in a future release.

hmb001, 2024-02-19:
I will follow your advice and set up a custom profile as my next step.

I would be very happy with the current setup save for **one** issue - the screen is currently only 77 characters wide. As shown in the attached image, the last 3 characters are cut off and the WHT color bar is cut off on the right side. If I set the line length correctly AND set the clock frequency to 14112000 then I get a lot of noise in the displayed characters.

[image]

IanSB, 2024-02-19:
@hmb001
>If I set the line length correctly AND set the clock frequency to 14112000 then I get a lot of noise in the displayed characters.
 
It does look OK with the Tandy 1000 defaults so maybe those defaults are already correct and the schematic I've been looking at isn't the correct one for your system.
Maybe you just need to adjust the H offset to centralise the capture area.

hmb001, 2024-02-22:
With the most recent stable firmware release I created a custom profile with the correct dot clock frequency, and display pixel width and height. The display is now perfect!

I have attached the profile file.
[Custom_Profile_0_.txt](https://github.com/hoglet67/RGBtoHDMI/files/14378048/Custom_Profile_0_.txt)

IanSB, 2024-02-22:
@hmb001
>I have attached the profile file. 

Thanks for that, I have put it in the "Other" folder named as "Zenith Z110" so it will be included in any future releases.

Did you find the create custom profile menu easy enough to use or does anything need clarifying?

hmb001, 2024-02-22:
The only issue that made me pause for a moment was the entry for "Pixel Height", until I realized that likely in most cases "Pixel Height" = "Number of scan lines". Also it seems "Pixel Height" must be an even number. The Z100 has 225 scan lines, so the entry for this parameter needs to be 226.

Am I correct in assuming the adapter can usually figure out H and V sync polarity? I didn't have to manually enter these parameters when creating the profile - or is that only the case when both H and V sync are positive?

IanSB, 2024-02-22:
@hmb001 
>Am I correct in assuming the adapter can usually figure out H and V sync polarity? I didn't have to manually enter these parameters when creating the profile - or is that only the case when both H and V sync are positive

Yes it auto detects and sets the sync polarity using the analyse timing option and can also usually work out if the source is composite or separate H/V sync.