hoglet67/RGBtoHDMI #368: SAM Coupe aspect ratio change?

G4DDS, 2024-03-02:
Just upgraded to latest stable release 20240213_e630d36f from previous stable release 20230517_eb620884. I have noted that on the SAM Coupe the screen no longer fills the whole monitor and as can be seen in the second screen shots the border is very small and not centred.

I seem to remember when SAM Coupe support was initially added it was like this before it was fixed to fill the whole screen and to centre it correctly. Is it broken again or am I confusing it with something else?

[image]
[image]

G4DDS, 2024-03-02:
I've found where this issue was originally fixed https://github.com/hoglet67/RGBtoHDMI/issues/226#issuecomment-901586484

IanSB, 2024-03-04:
@G4DDS 

I think this got broken when I made some changes to fix scaling issues with 4K monitors.
It should be possible to get it working again.

G4DDS, 2024-03-05:
@IanSB Excellent. I'll test the fix when its ready.

IanSB, 2024-03-09:
@G4DDS 
Can you try the attached files:

[Sam_Coupe.zip](https://github.com/hoglet67/RGBtoHDMI/files/14548097/Sam_Coupe.zip)

Replace both kernelrpi.img and the Sam coupe profile in \Profiles\6-12_BIT_RGB\Enterprise_+_Sam Coupe
with the files in the zip and reboot.
Delete any saved profiles as that will override the replacement profile.

IanSB, 2024-03-09:
@G4DDS 
Also try increasing the Max H Width setting in geometry up to a maxium of 640 (with the above new version) which may increase the border and help with centralising.

IanSB, 2024-03-10:
@G4DDS 

Thanks for testing but I've decided to make some additional changes as the current method for using the full width of the screen is a bit arbitrary and not easily testable across all combinations so I've made a specific option to do that instead.

The reason the original problem above happened was due to changes I made to avoid things like the PAL c64 scaling like this:
[image]
rather than this which is more accuratel:
[image]
However maybe some people would like the first option so I have added the "Integer Scaling (16:9)" setting in preferences as follows:

Normal 4:3 = use the centre 4:3 (ish) area only on widescreens
Use Full Width = use the full width of the screen area on widescreens where applicable while maintaing aspect ratio.
Full No Aspect =  use the full width of the screen area on widescreens where applicable and ignore the aspect ratio.

So I have set the Sam Coupe profile to default to "Use Full Width" although "Full No Aspect" should work as well but that might not always be the case with some combinations of screen and computer.

Here is the updated version:
[Sam_CoupeV2.zip](https://github.com/hoglet67/RGBtoHDMI/files/14550976/Sam_CoupeV2.zip)

This should behave the same as the previous although if you set "Integer Scaling (16:9)" back to "Normal 4:3" you will get the cropped image again.
Note you also have to update default.txt in /Profiles

G4DDS, 2024-03-10:
Just tested the latest fix and it behaves as you said it would. A coupe of points I noted:

1. FF OSD was set to OFF where the default is normally ON.
2. I have been playing a game called BOING today and I thought the scroller looked a little odd and based on the screenshots above (of the C64) I thought I'd try the game on SimCoupe to see what the intro looked like.

On the real SAM it looks like this:
[image]

But on SimCoupe it looks like this:
[image]

I've never seen (or if I have I have not paid any attention) anything use the top and bottom borders like this. I'm guessing it would be handy if the SAM display could be adjusted to switch between the current thin top and bottom borders or more representative bigger borders (while maintaining aspect ratio).

IanSB, 2024-03-11:
@G4DDS 
> I've never seen (or if I have I have not paid any attention) anything use the top and bottom borders like this. I'm guessing it would be handy if the SAM display could be adjusted to switch between [...]

You can only get exact aspect ratios using interpolated scaling which will soften the image. The very sharp integer scaling is always a compromise as you are restriced to exact multiples of the LCD pixels. 

The approximate pixel aspect ratio for various pixel clocks is as follows:
7 to 8Mhz = 1:1 (aka 4:4)
14 to 16 Mhz = 1:2 (aka 2:4)
The Sam Coupe uses a 12Mhz pixel clock which has a pixel aspect ratio roughly half way between the two above.
i.e. 3:4 but the current aspect ratio is set to 3:5 which crops off excessive borders top and bottom and fills the screen with more active video and is probably more appropriate for an NTSC source which has reduced top and bottom borders.

Here is another update which has some further preferences changes and also changes the profile to use 3:4 aspect ratio:
[SamCoupeV3.zip](https://github.com/hoglet67/RGBtoHDMI/files/14563655/SamCoupeV3.zip)

The preference menus are changed as follows:
16:9 Integer Scaling: options = Exact 4:3 / Enhanced 4:3 / Full Width

Exact 4:3 means use the exact centre 4:3 portion of 16:9 so for 1080p that would be 1440x1080
Enhanced 4:3 means use a wider centre section of 16:9 so for 1080p that would be 1600x1080 (default for most)
Full Width means use the full width of 16:9 so for 1080p that would be 1920x1080 (used by Sam Coupe)

The above profile now defaults to 3:4 which I think is more technically correct but to get back to the old 3:5 ratio use the swap aspect setting in preferences

G4DDS, 2024-03-12:
Screen captures with default SAM Coupe Full Width setting:

[image]
[image]
[image]

I think that's probably pretty close to how it should be.

G4DDS, 2024-03-24:
@IanSB I've switched my setup to the 3:5 ratio as the 3:4 seems a little too stretched horizontally for my liking. Still nice to be able to swap it at will if I notice a demo or game using the border to produce effects.