hoglet67/RGBtoHDMI #373: Red background with and without source

briskspirit, 2024-03-29:
Any idea why that happens? Have I killed CPLD again?

[image]
[image]

IanSB, 2024-03-30:
@briskspirit 
> Any idea why that happens? Have I killed CPLD again?

It's possible but it looks like alternate pixels are stuck on red is that correct?
If that's the case it isn't a damaged input but rather a problem with communication between between the CPLD and the Pi zero.
(I assume you have scanlines enabled as well which would result in the black lines between the lines of red/black pixels)

The 6 bit pixels are passed from the CPLD to the Pi two at a time over a 12 bit bus so it looks like one half of the bus has a stuck red data pin. It might be a damaged CPLD or Pi zero pin but it is more likely to be a bad solder joint on the CPLD or it might be a bad connection between the CPLD board and the Pi zero on the 40 way header.
There are two red bits per pixel so the following pins on the 40 way header are used to transfer red data:

1st pixel red hi = 40 way header pin 3 (Q00) CPLD pin 29
1st pixel red lo = 40 way header pin 29 (Q03) CPLD pin 5
2nd pixel red hi = 40 way header pin 24 (Q06) CPLD pin 12
2nd pixel red lo = 40 way header pin 23 (Q09) CPLD pin 13

Check the continuity of those pins between the CPLD board and the Pi zero.
Also check for a bad solder joint on the associated CPLD pins.
 
For refererence the 12 bit bus which is marked as Q00 to Q11 on the schematic is used as follows:
Q00 R0 HI
Q01 G0 HI
Q02 B0 HI
Q03 R0 LO
Q04 G0 LO
Q05 B0 LO
Q06 R1 HI
Q07 G1 HI
Q08 B1 HI
Q09 R1 LO
Q10 G1 LO
Q11 B1 LO

briskspirit, 2024-03-30:
Interlace isn't enabled and today (after refreshing? idk..) picture have changed to vertical lines.
[image]

Flashed to BBC - getting white tint. Flashed to YUV - blue-ish background(dark blue). 
Tested with all "test color" profiles (BBC, YUV, RGB) - everything looks good! 

[image]

Seems like if color test on all 3 FWs passes - it's not RP-to-CPLD problem.

So red got "stuck" on my buffer board with dsub9 connector? (Which was working fine initially before I put it into 3D printed case and added "UPS" to RP)

IanSB, 2024-03-30:
@briskspirit 
>Interlace isn't enabled and today (after refreshing? idk..) picture have changed to vertical lines.

That is still alternate pixels stuck on red

>Seems like if color test on all 3 FWs passes - it's not RP-to-CPLD problem.

The colour test is mainly meant for testing monitors, not the CPLD and it only tests the frame buffer of the Pi, not the CPLD to Pi communication.

Have you tested continuity of the above mentioned pins from the CPLD to the Pi zero?

> added "UPS" to RP

What do you mean by that?

briskspirit, 2024-03-30:
Continuity looks good, tried to re-solder Pi header as I clipped it (thought maybe I have cold joint).
I am kinda giving up with the result that it's faulty Pi Zero W , but everything is fine with Pi Zero 2W. Something broke on Pi side and after header. RIP Pi Zero )

Not sure if this issue should be closed or kept for other people as a reference @IanSB 

P.S. by adding "UPS" I meant that I added USB-C charger + 750mAh LiPo battery so I don't need to use USB power while tinkering with my old stuff.

IanSB, 2024-03-30:
@briskspirit 
>I am kinda giving up with the result that it's faulty Pi Zero W , but everything is fine with Pi Zero 2W

It looks like the CPLD is OK if it still works with the zero 2W. 

If you were using the Zero W with the previous CPLD that shorted and failed, perhaps that also weakened the Pi so it failed as well some time later.

Apart from soldering issues the only other problem I've seen occasionally with the 12 bit bus is intermittent contact on the 40 way header caused by flux wicking into the 40 way socket after soldering and that can be fixed by using iso propyl alcohol on the 40 way pins and socket.

>P.S. by adding "UPS" I meant that I added USB-C charger + 750mAh LiPo battery so I don't need to use USB power while tinkering with my old stuff.

Another option if your monitor has a built in USB hub is to power the Pi from that using a USB to micro USB cable so no separate power supply needed.