hoglet67/RGBtoHDMI #374: Shorted/faulty CPLD?

axmandm, 2024-04-03:
I have an RGBtoHDMI purchased back in 2022. When I tried to use it yesterday, I only got a green output from my BBC Master.

I thought perhaps there was an issue with the Q8/Q9/Q10 output transistors, but a scope showed similar output signals on each (i.e. colour information).

If I select Colour Test as the profile in the Main Menu, I just get a flashing screen (i.e. black, then back to the menu).

I have tried reprogramming the CPLD to RGB/YUV etc, this makes no difference.

Does this sound like a bad CPLD?

IanSB, 2024-04-03:
@axmandm 
>I only got a green output from my BBC Master.

Do you mean green text or just a blank green screen?
Is it a digital board with fixed cable or one with the analog extender and interchangeable cables (which can also be used with the BBC)?

>If I select Colour Test as the profile in the Main Menu, I just get a flashing screen (i.e. black, then back to the menu).
>I have tried reprogramming the CPLD to RGB/YUV etc, this makes no difference.

What CPLD was programmed when you got the flashing?
Does the CPLD programming appear to proceed and finish normally as expected? (Which means the CPLD is at least partially working)

>Does this sound like a bad CPLD?

Possibly, but it could be several other things too like monitor problems, corrupted software, broken wires in the DIN plug or the IDC connector getting unplugged from the 12 way header on the CPLD board.

First thing to try would be to make sure the HDMI cable was fully plugged in at both ends (or try another cable or even another monitor if you have one), then update the software to the latest release in case your existing SD card got corrupted in some way. (Reformat the SD card first, don't overwrite the existing files.)
Then check the connections in the 6 way DIN connector and the 12 way IDC connector on the other end of the cable (assuming a digital board).

It might be useful if you posted screencaps of the source summary page and the system summary page in the info menu (make sure a suitable BBC or Master profile is selected).
Press Up and Down simultaneously while in those pages to screencap the output with files saved to the SD card.

frank-f, 2025-01-02:
I have a kinda related question. In an act of interest and boredom I put together one RGBtoHDMI back in 2022 but somehow it wasn't working right away. I debugged it a bit, then got distracted, it went in a drawer and I forgot about it. Now recently while reorganizing things I found it again, hooked it up to a monitor and was instantly reminded, why it was in the drawer. The screen was blank.
It shows a menu and can program the CPLD programs without error if it is blank, but once it's programmed, it's like the thing is dead. Does something need to be hooked up to the inputs for the Menu to work? Or should I be able to use the menu regardless? Is there a way to test if the CPLD works?

IanSB, 2025-01-03:
@frank-f 

>Does something need to be hooked up to the inputs for the Menu to work? 

No, you should be able to call up the menus even with nothing connected but if one or more important CPLD pins aren't connected properly or are shorted you can get lockups like that.
Try holding down all 3 menu buttons while booting up to get into the CPLD recovery menu which should allow you to use the buttons. Then go to the info menu, use save log and edid and post log.txt from the SD card

frank-f, 2025-01-03:
Thanks for getting back to quickly. The recovery menu works fine, I used it a bunch of times to reprogram the CPLD already. Here's the log, I hope you can spot something wrong in it. :)
[Log.txt](https://github.com/user-attachments/files/18298320/Log.txt)

IanSB, 2025-01-03:
The log shows
CPLD  Design: Simple
CPLD Version: 0.0

This should show the type and version programmed into the CPLD

This means it thinks it is connected to a c0pperdragon style simple interface for the Amiga, not a CPLD and this is detected when the version pin (33 on the CPLD) is pulled low which can happen when using CPLDs already programmed with other unrelated firmware but might also happen if that pin is shorted to an adjacent pin.

>The supply situation was difficult in 2022 so I had ordered it from AliExpress...

CPLDs from that source usually work but are frequently already programmed with unrelated firmware which will stop you from reprogramming them with RGBtoHDMI firmware.

Are you sure the CPLD is being reprogrammed?
After starting the programming procedure you should see:
Identifying
Erasing
Programming
Verifying
Successful rebooting in 5 seconds
With the whole procedure taking about 20 seconds.
Anything else means it's not being reprogrammed.

If that is your problem, you have to cut the three solder jumpers on the underside of the PCB, JP1, JP2 & JP4, try reprogramming again and if successful remake the jumpers with solder bridges (You MUST remake the jumpers for it to work)
After that normal reprogramming should work without having to cut the jumpers.

On your photos it looks like there might be a short between the two pins arrowed which are 14 & 15 (although that would not cause the current issue) and there are shiny areas between other pins although they might all just be reflections of the light source.

[image]

If the problem isn't to do with the above rogue firmware, you need to check for shorts between each  adjacent pair of pins with a multimeter if possible, especially around pin 33.

frank-f, 2025-02-02:
I'm sorry that it took me this long, but while cutting the jumpers I had accidentally ripped some traces and never had the time (and motivation) to repair them. However, today was the day. After making the necessary connections I went into CPLD the recovery mode, programmed it, connected the jumpers again and ... it's still not working. Programming takes about 20 seconds, just like you described, but afterwards it still says Simple 0.0. I have really triple checked for loose or shorted pins, everything around the CPLD checks out, especially pin 33. Voltage drop between 33 and GND is 0.6V and between 33 and 32 it's 1.7V, so 33 is definitely not shorted. I have no idea anymore.