hoglet67/RGBtoHDMI #376: Audio Support

angelsix, 2024-04-23:
Starting to use this project in lots of builds and it is absolutely awesome. I'm amazed at the quality of the output, the feature richness, and overall skill gone into making this project so universal and professional. Hats off to all involved.

On to the issue. Everyone who gets these mods the first instant issue is lack of audio. Most people do this mod so they can connect up to a monitor, that has HDMI input, no 3.5mm or RCA audio input. 

As this project seems to lack any audio support, the only option is a £60 HDMI audio embedder device to re-inject audio into the HDMI.

Is there any way to add audio support to this project? I'm both a hardware and software dev so I can possibly assist to if you point me in the right direction. Perhaps sending analog audio into the Pi, or making a dedicated hardware board to inject audio into the HDMI pins directly perhaps? 

Basically, a solution cheaper than a £60 embedder device.

IanSB, 2024-04-23:
@angelsix 
>Is there any way to add audio support to this project?

I am looking at this but the main issue has been lack of documentation to output audio over HDMI from bare metal on the Pi but I have recently found some info that should help with that.
The other issue that might indirectly help with this is that the CPLD is being discontinued at the end of the year so I am currently redesigning the board to use an alternative solution and I may be able to add audio capture to this new design.

>Basically, a solution cheaper than a £60 embedder device.

There is a cheaper embedder available for about £35 which I have tried and does seem to work OK but I haven't done exhaustive tests: https://www.aliexpress.com/item/1005006695915253.html
(I didn't buy from this seller but that is currently the cheapest price. There are several other sellers listing it on that site)
It is also listed on ebay etc at a slightly higher price.

IanSB, 2024-05-03:
@angelsix 
I managed to get bare metal HDMI Audio output working and I have uploaded an Alpha test version with a HDMI audio test menu if you want to give it a try:
https://github.com/IanSB/RGBtoHDMI/issues/29

This is only half of the solution as I still need to devise a way to capture the audio in the first place but at least it is half way there!

It works with the TVs/monitors I have tested so far but others have reported problems with some TVs so let me know if it works for you.

You may get some clicks in the audio and that can be fixed by editing config.txt and changing:
disable_l2cache=0
However that change introduces video glitches in high bit depth video so some futher work on video capture buffering needs to be done before that change can be made permanent.

Also

angelsix, 2024-05-12:
I have copied the files to the SD card and overwrote the existing files that are there. Am I meant to do something more? No audio for anything but amiga boots as normal

IanSB, 2024-05-12:
@angelsix 
You can just play audio files from the menu
[image]

[image]

IanSB, 2024-05-12:
@angelsix 

If you can't get sound, check that your monitor or TV isn't configured for line input.
Sometimes even testing with another HDMI source isn't enough:

When I first tested this on my 12 TVs and monitors, it didn't work properly on four of them:
Two had technical issues which were caused by bugs that have been fixed in Alpha65G. (This fixed all problems reported by testers)
Two had no audio output and one of those was fixed with a simple menu change to disable line input.

The last one (an LG TV) was a bit more unusual. This had no audio output when running the test but worked when a Sky receiver was plugged into the same input. After much experimentation I found that it worked when the RGBtoHDMI output was configured for 50Hz but not at 60Hz which was why it worked with the Sky receiver which is 50Hz and that eventually turned out to be caused by labelling the input as "PC" in the TV's menu. Doing that meant when a 60Hz PC source was connected it used the line input and when a 50Hz source was connected it used the HDMI audio but wasn't obvious that making such a cosmetic change would change hardware behaviour.

I think some monitors will also override HDMI audio if something is plugged into the line input (if available)

angelsix, 2024-05-12:
yep works for me on all monitors

1stage, 2025-01-11:
IanSB, which CPLD are you looking to switch to?

Also, any further progress on audio?

IanSB, 2025-03-22:
@angelsix 
@1stage 

Coming soon:

[image]

1stage, 2025-03-22:
Woohoo! Let me know how I can help!

Minimally, I'll upload the Gerbers, BOM, and CPL files so other Aquarius users can access them.

1stage, 2025-03-30:
@IanSB, any early info on where the audio will be coming in? RasPi pins? CPLD pins?

IanSB, 2025-04-10:
@1stage 
@angelsix 

Some details of audio support here:
https://www.stardot.org.uk/forums/viewtopic.php?p=452912#p452912

1stage, 2025-04-11:
@IanSB, I replied on Stardot. 

Should I check your audio dev branch for an updated schematic?

Very exciting!

bixxewoscht, 2025-04-11:
@1stage I'd be happy to test your new audio board with my RGB2HDMI. I'm using it with the NES (PPU Digitizer) and audio was the only thing missing for me.

Feel free to message me on Github if I can test, I don't have an account on Stardot.
Thanks for your continuing developments!