hoglet67/RGBtoHDMI #377: Unwanted red scanlines (Mac Classic EU, 20240213, 6-12 Bit RGB)

ctshh, 2024-04-24:
First of all, and foremost - **thank you**! A wonderful project that is helping preserve computers I grew up with (or, wished for at that time, that is). So consider this "issue" a humble request for help by anyone interested.

I have a Macintosh Classic (EU / 230V version if that is of interest) hooked up to a RGB2HDMI, Software version 20240213, CPLD 6-12 Bit RGB v9.4, Interface 8/12 Bit Digital RGB(TTL) on a Pi Zero W.

After having had no luck with the built-in Mac Classic Profile (no picture at all) I worked my way through a CGA (Old), which showed a picture when the OSD was on, but did the flashing when the OSD was off, to the now working custom Profile. Which, interestingly, only works when I chose -H-V instead of the +H-V. Anyhow, heres the Profile that works:
[Custom_Profile_0_.txt](https://github.com/hoglet67/RGBtoHDMI/files/15090441/Custom_Profile_0_.txt)

The problem I am facing are (red?) scanlines that run across the whole screen*, from the very top to the very bottom. I have tried "anything"(tm) in the Sampling and Geometry section, to no avail. For a demonstration, see here:
https://youtu.be/VZypCfY9kp0

https://github.com/hoglet67/RGBtoHDMI/assets/6263/e72912c2-2b73-497d-8ced-e401c1c3ba96

Any ideas what could cause this and how to get rid of this very last bit of visual annoyance? And yes, the "virtual" scanlines feature of the RGB2HDMI is off.
Thanks!

*)Only on the Mac-Part, the black border does not show the scanlines.

IanSB, 2024-04-24:
@ctshh 

Looking at your profile txt file:
```
sampling=1,1,1,1,1,1,1,0,6,0,0,0,0,0,1,0,0,7,0,0,100,256,100,256,100,256,256,256
geometry=168,23,512,342,512,342,2,2,0,2,15667200,704,0,370,0,0,0
palette=Mono_(2_level)
ntsc_type=1
pal_oddline=0
teletext_deinterlace=0
output_colour=1
scanline_level=0
vsync_indicator=1
genlock_mode=1
powerup_message=0
```
These two settings are the cause which probably got changed from their defaults while you were experimenting:
```
vsync_indicator=1
genlock_mode=1
```

vsync_indicator=1 turns on the red bar (this is the "V Sync Indicator" option in the settings menu.
This is primarily a debug setting used for checking if the output HDMI signal is locked to the input video in which case the bar will be stationary.
Also genlock_mode=1 ("Genlock Mode" in settings) turns off the genlock which means the output is not locked to the input which is why your bar is moving.
First try changing genlock mode to "On (Locked)" in settings and the bar should stop moving
Then change V Sync Indicator to "Off" which should remove the bar
Finally save your modified configuration in the main menu. (Note this modified saved file will be in Saved_Profiles, not Profiles)

Or alternatively edit the above txt file to remove those two lines which will set them back to their defaults.

ctshh, 2024-04-24:
And with "just" that, I now have the most stable, clear and beautiful HDMI output of my beloved Mac Classic I could have ever  wished for.
Attached, for everyone else looking, my (!) working custom config for a Macintosh Classic.
[image]
[Custom_Profile_0_.txt](https://github.com/hoglet67/RGBtoHDMI/files/15099558/Custom_Profile_0_.txt)

closed 2024-04-24 by ctshh