hoglet67/RGBtoHDMI #378: YUV modes without analog board

luisfcorreia, 2024-05-01:
Hi,

is it possible to make the board work with YUV signals from a ZX Spectrum directly to a 12bit board without using the analog add on board?

I might improve signals coming from ULA to better align with TTL.

But my question is if the CPLD code disregards YUV mode by the analog board not being present.

IanSB, 2024-05-06:
@luisfcorreia 

I assume you are talking about the original 48K Spectrum as the later ones are RGB?

The YUV in the Spectrum is analog out of the ULA so there are no digital YUV signals to pick up.

However there is now a lumacode solution for the 48K Spectrum so you can get full colours out of the Spectrum with an analog RGBtoHDMI and a c0pperdragon lumacode board (You can re-use the modulator phono to send the lumacode signal out of the Spectrum)
For more details see here:
https://github.com/c0pperdragon/LumaCode/wiki

luisfcorreia, 2024-05-06:
yes, the original 48K

I see, analog only signals

I was tempted to build also the Analog board extension boards, specifically for this, would that work?

(I'm not really interested in Lumacode based solutions)

IanSB, 2024-05-18:
@luisfcorreia 
Spectrum 48K support is very marginal so not guaranteed to work due to the noisy YUV signals and if it does work you don't get bright colours anyway. It is worth experimenting with it if you already have an analog board for other computers but I'm not sure I would build one just to try it.
At the moment the only way to get full RGB support with an analog interface is to use one of the following:

1.  [c0pperdragon's open source Spectrum 48K component interface](https://github.com/c0pperdragon/ZX-Spectrum-Component-Video)

2. The VLA82S+ ULA replacement with RGB output option

3. A lumacode adapter