hoglet67/RGBtoHDMI #379: Which bus line is broken?

venice1200, 2024-05-05:
Hi,
at first, many thanks for this absolut great Project.
I use is for my Amiga 500 and C64.

Actually I build a PCB for the A500 from Solarmon but can't get the CPLD programmed.
https://github.com/solarmon
```
RGB to HDMI booted
Running GPU capture build. Kernel version: 20240213 e630d36f
Marked framebuffer from 1EB00000 to 1EC00000 as cached
Pi Hardware detected as type 3
CPLD board detected
8/12 bit board detected
Standard board detected
Using PLLA as the sampling clock
PLLA: 2400.000000 ANA1 = 00000001
...
EMMC: BCM2708 controller power-cycled
Unknown CPLD: identifier = dcf
Keycount = 0
CPLD  Design: 3-12_BIT_BBC
CPLD Version: c.f
Loading file /Palettes/Atari_2600_NTSC.bin
```
I have double checked each line/connection but can't find any problems.
Maybe someone can tell me which one of the 12 bus line(s) is the problematic one.

Full log attached.

Cheers

IanSB, 2024-05-05:
@venice1200 

```
Unknown CPLD: identifier = dcf
```

The CPLD is already programmed with some unknown code and that will stop initial reprogramming with RGBtoHDMI code.
The reprogramming GPIOs are shared with other pins and if the unknown firmware makes those pins outputs it will conflict with the reprogramming sequence. The fix on a normal CPLD board is to cut JP1, JP2 and JP4 to isolate the reprogramming lines, then program the CPLD, then remake those links. If those solder jumpers are on the equivalent Solarmon board then cut them otherwise you will have to cut the relevant tracks.

venice1200, 2024-05-06:
Hi @IanSB,
I have heard about already programmed CPLD.

I will see what I can do. 

Many Thanks

closed 2024-05-06 by venice1200