hoglet67/RGBtoHDMI #380: New RaspberryPi Revisions need newer Firmware

explit7, 2024-05-07:
Hello @hoglet67 I tried to use your RGBtoHDMI with a newer Revision of RaspberryPi 3A+ and found out, that, the newer Firmware (which was needed to boot this RPI) was not included with you latest release.
I googled around and found exact the same Problem with batocera Release: [https://retropie.org.uk/forum/topic/34840/brand-new-raspberry-pi3a-wont-boot-with-any-version-of-retropie-ive-tried-solved](url)

The solution was very simple:

- Download latest Raspberry Pi Firmware Files: [https://github.com/raspberrypi/firmware/tree/stable/boot](url)

- Copy all **start*.elf** and **fixup*.dat** files to the SD-Card.

Now i can confirm its working fine with my newest RPI 3A+.
Please include this file with the you next release.

Thanks

IanSB, 2024-05-08:
@explit7 

Can you give the latest Alpha 65G release on my fork a test:
https://github.com/IanSB/RGBtoHDMI/issues/29

I updated the firmware a few weeks ago.
Also can you check the HDMI audio output test works on your monitor(s).

explit7, 2024-05-09:
Hello @IanSB, I tested your fork. Also  on your fork, new firmware files are missing. Your fork works after i copied new Raspberry Pi Firmware onto SD-Card. I couldnt test HDMI Audio, because on my Micro8088 IBM PC/XT i have an old Nokia VGA Monitor connected.

Here are the Firmware file which are needed for the new Raspberry Pi 3A+ (and maybe other newer Revisions)
[Uploading new_firmware.zip…]()

IanSB, 2024-05-09:
@explit7 
Are you sure you tested the right version? (Alpha 65G posted mid-way through that issues page, not the file at the top which I have now removed)
I just downloaded the Raspberry Pi Firmware Files from the above link again and they match the ones I downloaded on the 23rd April and they are different and more recent than the ones in your zip which are datestamped in March.

explit7, 2024-05-09:
I tested Alpha65G from here: https://github.com/IanSB/RGBtoHDMI/files/15243056/Alpha65G.zip 
The new Firmware files are also missing.

Please check, the new Firmware files from this archive 
https://github.com/hoglet67/RGBtoHDMI/files/15257014/new_firmware.zip
need to be on the root of the SD-Card, otherwise the new Revision RaspberryPi 3A+ will not boot at all.
I know, this is very strange, but i checked to Raspberry Pi 3A+. One old Revision - works like a charm out of the box, and the new Revision (which look the same) but doesn't boot at all, if the new firmware files are not on the card.

IanSB, 2024-05-10:
@explit7 

Only one pair of firmware files should be needed so I'm trying to work out what is going on here.

There are 4 pairs of firmware:
start.elf + fixup.dat - the standard firmware
start_x.elf + fixup_x.dat - the extended firmware
start_cd.elf + fixup_cd.dat - the cut down firmware
start_db.elf + fixup_db.dat - the debug firmware
These are for the Pi zero, 1,2 & 3
The versions with 4 in the filename are for the Pi4 etc.

This project loads the cut down ( _cd ) version in the config.txt which is why only that pair are on the sd card
These files are more recent than yours so I don't understand what is going on unless one set of firmware is loading another in some circumstances

On your working SD card can you remove all the files except the two _cd versions and reboot
If that stops working, add the following pairs back in order and reboot after each pair until it works again:
start.elf + fixup.dat 
start_x.elf + fixup_x.dat
start_db.elf + fixup_db.dat
all the 4 versions

explit7, 2024-05-10:
@IanSB You are right. My mistake. Somehow, i messed something up with different SD-Cards.

Your Alpha65G Build works OOB on my newer RPI3A+.
This is enough, that this files from your alpha65G Firmware are placed on the sd-card
fixup4cd.dat
fixup_cd.dat
start4cd.elf
start_cd.elf

I created an archive with this 4 files (in your newer versions)
[RPI_NewFW.zip](https://github.com/hoglet67/RGBtoHDMI/files/15279643/RPI_NewFW.zip)

I will change the thread a little bit, to not confuse other users.