hoglet67/RGBtoHDMI #381: C128 80 columns mode via "Analog board"

c0pperdragon, 2024-05-08:
Disclamier: I don't have a C128, so this idea is purely theoretical.

I believe it possible to wire up the 9-pin video port to be used with the 6-pin Analog inputs. This would be very convenient for C128 users as only one RGBtoHDMI is then needed for both output options (80 columns, 40 columns Lumacode). I could also imagine a Y-cable with a switch for maximum convenience. If there would also be a possibility for the RGBtoHDMI to auto-switch profiles this would be nice.

The cable would just use some resistors to combine the digital RGBI signals to get RG, GB, GI to go into R, B, G respectively. The R and B inputs would need to distinguish between 3 levels and the G would need do detect 4 levels. The SYNC input directly uses the monochrome output from the C128.

@IanSB I don't know if you have a C128 in your collection. If yes, I could build such a cable to give this a test. If this works, I would make a new product from this.

IanSB, 2024-05-08:
@c0pperdragon 

Yes that should work with an appropriate profile and I do have a C128

c0pperdragon, 2024-05-09:
I changed my mind a bit here. Instead of such a C128-specific cable, I think it is better to make a generic CGA/C128 cable to feed into the Analog board. 
For the sync signal, using a resistor (from HSYNC) with a diode to VSYNC should get a combined sync that will be low for several scanlines to denote vertical sync (same what the ZX Spectrum does natively). I guess the RGBtoHDMI can handle this.

This passive circuitry should be suitable for other 4-bit TTL signals as well. Many users may be able to solder up their own cables (or adapters to DB9) when provided with schematics. This would increase the range of applications for the RGBtoHDMI Analog to a point where it really makes sense to have a dedicated integrated board (like the RGBtoHDMI Mono & Lumacode) for analog inputs only.

I am not sure about the additional comparators on the R and B lines, as the 4-level detection is very rarely needed. Leaving this out would keep costs down a bit and it may also be necessary so to fit everything on the board.
When the IDC-socket is placed correctly, this would also directly fit the existing cases.  :-)

IanSB, 2024-05-09:
@c0pperdragon 
>For the sync signal, using a resistor (from HSYNC) with a diode to VSYNC should get a combined sync

There is a 1K pulldown on the sync input so I have previously used two diodes to OR H and V sync but that will only work with +ve going syncs.

c0pperdragon, 2024-05-09:
To create a combined sync that can be interpreted by the RGBtoHDMI, it is probably enough to use two resistors. A smaller resistor (lets say 1K) for hsync and a larger one (2k) for hync. These values are large enough to not strain the output too much, but it should allow quite fast signal propagation. This wiring will form a simple 2bit DAC and I believe by choosing the correct compare point and polarity, this should work like an 'OR' or 'AND'.

The same 2-bit DAC can be used to combine G and I to feed into the G line. I will also add an external 1K pulldown resistor here to get the voltages into the detectable range. The R and B lines can be taken directly.

c0pperdragon, 2024-05-10:
Maybe I will turn the whole solution on its head. My real objective here is to provide a solution to use both output modes (native RGBI or lumacode) of the C128 with a single RGBtoHDMI. But instead of forcing the RGBI signal through the 3-pin analog inputs, maybe it is better to decode the lumacode into digital signals instead to use a digital RGBtoHDMI.
 
So my current vision is now a small device that plugs into the DB9 output and passes most of the signals straight through to the RGBtoHDMI. It also draws power from the RGBtoHDMI as it is not available otherwise. This device has also a short cable with an RCA plug to fetch the lumacode from the modified RF output. The individual lumacode samples are provided on 2 extra data lines to the RGBtoHDMI. The sync signals are either forwared from the VDC or the VICIIdizer and sent on hsync/vsync (vsync being idle in case of lumacode) of the RGBtoHDMI. The RGBtoHDMI should be able to decide which input is present from this sync signal and switch to the necessary profile.
A very important aspect of this solution is the auto-switching capability. Do you know if there is a reliable way to know from looking at the RGBI port alone if the C128 currently runs in 80 columns mode?

IanSB, 2024-05-10:
>A very important aspect of this solution is the auto-switching capability. Do you know if there is a reliable way to know from looking at the RGBI port alone if the C128 currently runs in 80 columns mode?

No, I don't think there is any easy way to detect as both outputs continuously generate sync.
(There are also some programs that make use of both screens)

>The individual lumacode samples are provided on 2 extra data lines to the RGBtoHDMI.

That wouldn't work as the data would have to be fed to the same pins as the 80 column mode plus you would need to switch the sync.

This seems overcomplicated compared to a simple passive adapter that could be switched with an RCA switchbox and would still have to be manually switched.

c0pperdragon, 2024-05-11:
Without auto-switching this would of course be overengineered for no good reason.

Still I would like to come up with some better solution for the C128. Particularly needing two RGBtoHDMI's is a bit cumbersome. Of course if the user wants two screens at the same time, there is no way around it, but most of the time this will not be the case.
The previous idea forces RGBIHV into RGBS for the Analog board. With a simple double-pole / double-throw switch the G and the S line could be switched between inputs.  Maybe I will indeed go with this.
Just to make sure: Do you think the RGBtoHDMI will be able to auto-switch between both inputs. The distinguishing feature will then be the fact that the or-combined csync from the VDC will have a solid vsync pulse that lasts several lines.

jdryyz, 2024-05-14:
👍 I see I did not have to make my own post on this subject. RGBI from a C128 to RG2HDMI was available for a while. So when Lumacode was announced for the 40 col side, I immediately thought all we need now is a way to get both signals into the same RGB2HDMI board and have it sense which is one is being selected.

Would be very cool if this could happen. I understand some but not all the challenges involved. Is it true that a C128's standard A/V and RGBI have active sync at the same time?  

I'm just imagining being able to toggle between modes using the 40/80 col switch alone.   :)

c0pperdragon, 2024-05-14:
@jdryyz  Don't set you expectations to high. It is still not possible to squeeze the VDC output into the single lumacode-line for use with the cheap lumacode RGBtoHDMI. You will have to use an RGBtoHDMI with "Analog board" that has the 6-pin connector (3 color lines, 1 sync, GND, +5V). 
But if this goes as I intend, you only need to toggle the switch on the Y-cable. 

I am now thinking about the physical appearance of this cable. Which of the following options would you think is most convenient?

1. Female DB9 connector with the switch and one RCA-cable going to the lumacode output and one ribbon cable going to the RGBtoHDMI.
2. The switch and some electronics is on a small adapter that plugs directly into the RGBtoHDMI and has two cables to the DB9-connector and the lumacode output respectively.
3. Like 2, but only with sockets to connect your own RCA and DB9 cable of any length you like.

jdryyz, 2024-05-14:
Agreed my expectations are a bit lofty. I was even thinking the electronics to pull this off would be small enough in cram *inside*, making all the necessary connections internally with a single HDMI connector sticking out somewhere.  :) 

In all seriousness, I prefer any solution with the least about of bulk.

In the mean time, I will have to fetch me another Pi Zero.

c0pperdragon, 2024-05-14:
I guess, I just found a solution that is most versatile:
Having just a cable for the RGBI output to go to a 6-pin "Analog board". This uses a few resistors that can be easily fitted into the DB9 connector housing. This cable can be used for the C128 as well as other machine with CGA output.
In my store there is already a cable to bring lumacode to the 6-pin "Analog board". So with manual swapping it is then already possible to use the same RGBtoHDMI for both output variants of the C128.
To bring some convenience to this, I will make a simple switchbox that plugs directly into the Analog board and provides two 6-pin inputs to be manually switched over. This switchbox can also come handy if you want to wire up any two machines to a single RGBtoHDMI.

jdryyz, 2024-05-14:
Excellent!

c0pperdragon, 2024-05-16:
After much pondering, I changed my mind again. Now I believe the best solution for the C128 is actually a VDCdigitizer. This would be an interposer board pretty similar to the VICIIdizer and it would also generate a lumacode signal to be decoded by the RGBtoHDMI. For quite some time I thought the VDC output could not be squeezed into the single lumacode wire, but now I think it can be done with a bit of data compression.
For maxium convenience the VDCdigitizer will have a connector to plug the VICIIdizer into it and the signal from there gets directly forwarded  when the VDCdigitizer is not doing anything. So a single RGBtoHDMI (either Mono & Lumacode or Analog) is necessary and everything is auto-switched.

Please follow over to the new discussion thread if you want to participate in this endeavor:
https://github.com/c0pperdragon/LumaCode/issues/19

c0pperdragon, 2024-05-19:
With the VDCdigitizer out of the picture, the previous solution (cable for CGA/C128 to 6-pin Analog) is the most promising again.
To not lose my specializaion on lumacode I will not build such a cable myself, but I will share my ideas so you can build a cable yourself.
@jdryyz do you have the necessary soldering equipment/skills to try this? It would require to use a standard male DB9 connector and some 1K and 2.2K resistors and a 6-pin ribbon cable and IDC plug.

jdryyz, 2024-05-19:
I have the skills but lacking the time right now. I will be engaged in a minor home renovation that I expect to last a month or so.

I can recommend two YouTubers that would be good candidates, however:

Aaron at the Retro Hack Shack:

https://www.youtube.com/@RetroHackShack

Or Mark from the RetroChannel:

https://www.youtube.com/@TheRetroChannel

Both have featured Lumacode in recent videos, and both are C128 users/modders.

timmcross, 2024-05-19:
I would love to assist in testing.  I have made several RGB2HDMI boards but haven't tried a lumacode board yet.  I have a NTSC C128 and would love to have a solution like this.  Should I order some lumacode boards from lscsc?

c0pperdragon, 2024-05-19:
Very nice!
But you will not need a lumacode board, but a fully-featured Analog board (sandwiched under a digital RGBtoHDMI) for my idea.
Also it is completely theoretical still and maybe you need some support from Ian also to make the software work (but probably it can be done with no programming and only by creating a profile).

So my idea was to force the 6 digital signals from the VDC through the 4 inputs pins of the Analog board by combining certain signals with resistors. The RGBtoHDMI should be able to seperate them out again after some voltage calibration.
If you want, I can write up a detailed wiring diagram, but maybe you can also figure out what I meant:
Combine the G signal with 1K resistor and the I signal with a 2.2K resistor and a third 1K ohm resistor to GND and feed this into the G input on the Analog board.
Combine the HSYNC and VSYNC signal with a 1K resistor each and feed this into the SYNC input on the Analog board.
Connect the R and B signals straight to the R and B inputs.

The new sync can be detected as an OR-combination of HSYNC and VSYNC when the threshold level is set correctly. 
The combined G signal will have 4 different voltage levels that should be fairly easily seperable.

For making a switch box for this 4-line signal and the single line lumacode signal, you will only need one double-pole-double-throw switch.

timmcross, 2024-05-19:
Ah, ok.  I have built analog boards as well.  I can see where I stand with parts for those right now.  I will look through your post again and see if I understand it.

IanSB, 2024-05-19:
@timmcross 
@jdryyz 

This will also require some RGBtoHDMI software support for auto profile switching but it may be some time before that happens as I am currently working on audio support although most of the difficult parts of that have now been solved.

c0pperdragon, 2024-05-20:
When the solution should be so specialized that it will only need to support the combination of the C128's DB9 output with lumacode from the VICIIdizer (and not with arbitrary CGA cards), you can utilize the monochrome output signal that is already provided on the DB9. This can be directly fed into the sync input on the Analog board. So you save 3 resistors here.

c0pperdragon, 2024-05-20:
I made a wiring diagram for a minimal solution. When switching to the RGBI inputs, the monochrome pin is used for the sync signal. As far as I see in the C128 schematic this seems to be an XOR-combination of HSYNC and VSYNC. So the RGBtoHDMI may be able to detect this and provide auto-switching between this and the Lumacode signal which uses different sync pulses. 

[c128adapter.pdf](https://github.com/hoglet67/RGBtoHDMI/files/15380094/c128adapter.pdf)

idun-project, 2024-05-20:
I started designing a solution for converting and auto-switching the c128's two video modes a few weeks back, and it also includes LumaCode support. So you can use it to switch to an analog CRT/monitor, or to switch between LumaCode and the VDC, which the circuit converts to RGBS. It also provides 64KB VRAM for the VDC.

open-source hardware project here:  https://github.com/idun-project/r2v2

c0pperdragon, 2024-08-12:
I decided to not pursue this idea any longer. Using the digital VDC output to somehow merge 4 bits into a 3-channel RGB for use with an Analog board ist just not worth it. It is just better to use a digital RGBtoHDMI for this.

closed 2024-08-12 by c0pperdragon