hoglet67/RGBtoHDMI #382: A500 - Flashing pixels in some areas

TomBonk, 2024-05-17:
I recently bought an Amiga 500 with RGB2HDMI installed. For my part, I just bought an Rpi Zero and installed the system on an SD card.
The problem is the flashing pixels.
It seems to occur only in the pixel area of specific colors.
If I change the screen to 4 colors, nothing flashes. If I change the screen palette is also ok. In games, nothing is flashing.
I am currently using Workbench 640x256 8-color and MagicWB palette.

In the video you can see the problem.
[https://youtu.be/Fz__n81wwdQ](https://youtu.be/Fz__n81wwdQ)

Cheers

IanSB, 2024-05-18:
This is a known problem with the c0pperdragon simple (non CPLD) adapter and happens on some motherboards. 
If that is the type of board fitted, the only option with such motherboards is to use the CPLD version of the adapter instead as that has software adjustable sample timing and can compensate for the problem.

TomBonk, 2024-05-18:
Indeed, I have an early version of RGB2HDMI. In that case, I will try the CPLD version. From what I've already seen it's worth any money :)

Greetings and thank you.

closed 2024-05-18 by TomBonk