hoglet67/RGBtoHDMI #383: Pi 4 doesn't work with Amiga adapters

tyristori, 2024-05-31:
Latest 20240213 software. This message is displayed with c0pperdragon's Amiga adapter and with LinuxJedi's CPLD adapter:

[image]

CPLD is already flashed on CPLD adapter and works fine with Zero 1.

Doesn't make a difference if single_button_mode is enabled or not.

I've got Pi 4 to work only with the 12-bit adapter.

IanSB, 2024-06-02:
@tyristori 
>I've got Pi 4 to work only with the 12-bit adapter.

Not entirely surprising as it is only tested with that.
The most likely explanation is that either the internal pullups on the GPIOs are not being set correctly for the Pi 4 or they are too weak so it looks like the switches are all pressed. The normal CPLD board has external 1K pullup resistors on those switches.

Try fitting 1K pullup resistors on the three switch inputs which are on pins 35, 36 & 37 of the 40 way connector.  Connect the resistors to 3.3v which is on pin 1 or pin 17.

tyristori, 2024-06-02:
@IanSB

I added the resistors and the button is working fine now with Pi4.

IanSB, 2024-06-03:
@tyristori 
>I added the resistors and the button is working fine now with Pi4.

It might eventually be fixable in software but that is a good workaround for now as I'm busy getting audio capture working at the moment. I think all the difficult parts of that have been solved so hopefully I will get that finished soon.