hoglet67/RGBtoHDMI #384: Custom boards for various RGB outputs

angelsix, 2024-06-05:
I would love to start making custom projects for many consoles. Is there anywhere I can read up on the general process required to get RGB from various consoles into the RGB2HDMI boards to work?

For example, Master System. It outputs native RGB already. Perfect starting point to integrate.

c0pperdragon, 2024-06-12:
From theoretical considerations, the RGBtoHDMI with added Analog board sould be able to do this. Making a suitable profile and palette file is probably a bit involved. The project wiki should contain most of what is needed here.

angelsix, 2024-06-13:
Ok thanks, I'll try and delve in when I can to add some custom support. I'll take a look at the wiki

closed 2024-06-13 by angelsix