hoglet67/RGBtoHDMI #387: ZX81 with VLA81 display problems

G4DDS, 2024-06-21:
I previously had a ZX81 that composite modded and using this cable https://github.com/hoglet67/RGBtoHDMI/wiki/Cables#apple-ii--zx80--zx81--uk101--nascom--superboard-ii-or-any-other-mono-computer the RGBtoHDMI interface was fine.

I have installed this ULA https://www.vretrodesign.com/products/vla81-zx81-ula-replacement and removed the composite mod (as told to by the developer of the VLA81) which gives a stunning picture on my composite connected TV. But, when connected to an RGBtoHDMI interface it now just gives this:

[image]

This is with the VLA81 set so that its video mode switch SW1 OFF which is the setting the developer told me to use (he refers to this mode in his instructions as "1Vp-p into 75Ohm"). Switch SW2 enables or disables the back porch  signal and it makes no difference what position this switch is in.

Incidentally, if I put the video mode switch SW1 ON (referred to as "Extended Sync level"), I get this on both my composite TV and RGBtoHDMI interface:

[image]

I have rolled the software back to release 20230517_eb620884 as that was the last time I used it successfully with the old setup just to rule out any oddities with release 20240213_e630d36f but it behaves the same on both. It is quite possible that the VLA81 just needs some setup tweaks but I don't know where I should start.

Any thoughts?

IanSB, 2024-06-21:
@G4DDS 

You probably just need to adjust the DAC levels in the sampling menu to match the video levels being output by the VLA81

Select the existing "ZX81 Composite" profile and then go into the sampling menu and adjust the DAC-A Y Hi value up or down until you get a clean image. If the video isn't synced you will also need to adjust the DAC-E Sync level.

Once you have some working settings, you will need to run the auto calibration as well but to get the optimum DAC values, adjust them up and down until the image starts breaking up and set your final value to half way between those upper and lower values.

If you get a good profile, please post it and I will add it to the next release. (Will be in /saved_values on the SD card)

G4DDS, 2024-06-22:
It turned out this was a lot simpler than I imagined. After having a play with some of the settings it turned out all I needed to do was to change “75R Termination” from On to Off.

G4DDS, 2024-06-30:
@IanSB 

Perhaps in a future release you could add a specific profile (called "ZX81 vLA81") which is based on the Composite profile but with 75R Termination off.

IanSB, 2024-07-01:
@G4DDS 
I have done that already.

G4DDS, 2024-07-01:
@IanSB 

Nice. Is that available in a latest beta? I've had a look around and I can't seem to find it.

IanSB, 2024-07-01:
@G4DDS 
Not yet, but there will be a new beta soon on my fork