hoglet67/RGBtoHDMI #388: Olivetti M19 - feature request

jdelange, 2024-06-24:
Hi, excellent project to allow us to keep using our vintage computers! I am one of a small group of Olivetti M19 (1986) owners that are working hard to keep these nice machines alive.

The Olivetti M19 comes with CGA capability, but the frequency is lower than normal. This prevents us to hook-up any standard (multisync) monitor and there are very few Olivetti color monitors on the market. So this puts us in a difficult position.

One of the members on the [Vintage Federation Computer Forum](https://forum.vcfed.org/index.php?threads/paradise-pvc2-questions-on-olivetti-m19.1245902/page-5#post-1391695) suggested we would ask if your project could add the Olivetti M19 as one of the platforms to be supported.

I know this may be a long shot and the fan base of the Olivetti M19 is not huge, but I thought I'd give it a try anyway.

IanSB, 2024-06-24:
@jdelange 

If it has a normal CGA connector then it should just plug straight in and it can almost certainly be made to work with RGBtoHDMI as all timing is software defined using profiles and the most recent release has a create custom profile menu to aid with setting that up.

see:
https://github.com/hoglet67/RGBtoHDMI/wiki/Tutorial-on-Adding-a-New-Profile

Do you have any technical details of the M19 such as pixel clock frequency, line frequency, frames per second etc or a link to a schematic?

jdelange, 2024-06-24:
@IanSB  Thanks for picking this up so fast! Yes, it has a standard RGB connector. Good to hear that you think the timing can be fixed in software. I will read through your tutorial.
On the tech details: forum member Zare has offered to provide the technical specs. I will refer him to your post.

goatM19, 2024-06-27:
Hello,

I can provide measurement data for video signal or anything else. Machine has mono and colour configuration - I would concentrate on colour mode which is Plantronics Colorplus-alike CGA for the start.

The highest pixel resolution is 640x200. There are two clocks lines in the chipset, 4.77MHz and 14.31818MHz. https://theretroweb.com/chipset/documentation/faraday-xt-controller-fe2010-654f50e036c61834700220.pdf.

[image]
[image]
[image]

The HSync seems a bit faster than usual CGA.

goatM19, 2024-06-27:
> @jdelange
>
> If it has a normal CGA connector then it should just plug straight in and it can almost certainly be made to work with RGBtoHDMI as all timing is software defined using profiles and the [...]

Hello Ian,
Do you think we could buy the MCE2HDMI device, autosense the HSync rate and enter pixel resolution and clock in config? Does this have high chance of working considering the signals displayed above?

goatM19, 2024-06-27:
> If it has a normal CGA connector then it should just plug straight in and it can almost certainly be made to work with RGBtoHDMI as all timing is software defined using profiles and the most recent [...]

Hello Ian,
Does the MCE2HDMI device have high possibility of working with this since we can enter the clock and pixel res, and the different hsync should be autosensed?

IanSB, 2024-06-28:
>Does the MCE2HDMI device have high possibility of working with this since we can enter the clock and pixel res, and the different hsync should be autosensed?

Sorry, I don't know anything about the MCE2HDMI and only support the authorised sellers listed in the wiki. Those sellers provide financial donations from their sales which has allowed me to continue development over the past few years but if you want to use boards from other sellers that don't contribute to the project then you must contact those sellers for any support.

jdelange, 2024-06-28:
@IanSB 

Thanks for your feedback. Interesting to understand how your efforts are funded. I think there may be a misunderstanding here. I do not think @goatM19 suggested he would be sourcing the boards from  different seller. I for sure would by from a listed seller to support your efforts. I think what @goatM19 is trying to establish is if the current device (from a listed) seller can be made to work straight out off the box without major software changes as you suggested in your first comment.

So he is trying to figure out how risky the investment will be.

So given the data provided on the frequencies of the M19, does that change your confidence that the device will work on the Olivetti M19?

goatM19, 2024-06-28:
@jdelange 

Agreed. I'm not aware of the range of supported devices. @IanSB let me rephrase the question into whether official devices from your wiki would allow simple profile change to accept the signal depicted above.

Your help is much appreciated, a sight of this computer running without the original monitor is very rare. In the end our goal is to make that happen, we think your software is the best candidate for it and regarding hardware we'll try to get whatever you support officially.

IanSB, 2024-06-29:
@goatM19 
>let me rephrase the question into whether official devices from your wiki would allow simple profile change to accept the signal depicted above.

As long as the RGBI and H/Vsync are normal TTL level signals which appears to be the case from your waveforms then it should work with the standard PC version of RGBtoHDMI as the timing can be set using the create custom profile menu.