hoglet67/RGBtoHDMI #389: How to build

mwedmark, 2024-06-26:
I can't seem to find the files to order VIC-ii Dizer so that part of the C64=>HDMI puzzle is unsolved. 
Is it open to build using gerbers/kicad or only on order?

IanSB, 2024-06-29:
@mwedmark 
The lumacode boards like VIC-ii Dizer are not part of this project and are currently only available assembled. See:
http://github.com/c0pperdragon/lumacode
RGBtoHDMI just supports the lumacode protocol to decode the signals output by the above. 

However, gerbers are available for c0pperdragon's earlier YUV/RGB C64 adapter and that is also supported by RGBtoHDMI and produces an identical image quality to the lumacode adapter when used with an analog RGBtoHDMI.
https://github.com/c0pperdragon/C64-Video-Enhancement

mwedmark, 2024-07-03:
OK great. Then I understand, thanks!

closed 2024-07-03 by mwedmark