hoglet67/RGBtoHDMI #391: ZX 48k YUV + analog board

luisfcorreia, 2024-07-25:
Hi,

I've just completed assembly of two Analog v3 boards and have them hooked up to my ZX48k Issue 3b

Just asking if this is "normal" and what can I change to make it work reliably.

Image is the keyboard tester program of DiagCart 

[image]

luisfcorreia, 2024-07-25:
It needs a change in pallete and probably some other stuff, closing as not a problem

luisfcorreia, 2024-07-25:
it's getting better but having U and V connected make no difference, this is the best I can get out of it

[image]

IanSB, 2024-07-25:
@luisfcorreia 
Don't mess with the palette or other settings as that will definitely break things.
The only things that could need adjustment are the DAC settings in the sampling menu and then only by small amounts.

The alternating lines in the first image might indicate the U and V are swapped or that all of YUV are connected to the wrong pins so check that first.

It looks like you are running spectrum diagnostics so the best image to display for testing this is the test card as described here:
https://github.com/brendanalford/zx-diagnostics/wiki/Firmware#testcard

If you think you have connected everything correctly, restore the default config then temporarily change the palette from "Spectrum_48K_9Col" to each of the following palettes in turn and screencap the results:

"Test_4_Lvl_G_or_Y",
"Test_4_Lvl_B_or_U",
"Test_4_Lvl_R_or_V",

Post the above screencaps

luisfcorreia, 2024-07-26:
hey, got good and maybe not so good news

luisfcorreia, 2024-07-26:
However, in my crappy LCD composite video monitor, I get this:
[image]

Please disregard the pixelation and focus on the gradient difference on the color bars.

What I'm getting right now is a full blown color bar with no difference at all

luisfcorreia, 2024-07-26:
About the [wiki](https://github.com/hoglet67/RGBtoHDMI/wiki/Cables#48k-spectrum-analog-yuv)

These are the pinouts that "work for me"TM.

## 48K Spectrum (Analog YUV)
| RGBtoHDMI 6 way IDC |
| --------- |
| Pin 1  GND |
| Pin 2  SYNC |
| Pin 3  U |
| Pin 4  V |
| Pin 5  Y |
| Pin 6  +5V|

luisfcorreia, 2024-07-26:
All this is with a ZX 48K issue 3b
Trying now with an Issue 2, my original machine from 1982
Currently I'm getting no sync detected

IanSB, 2024-07-26:
@luisfcorreia 
>Please disregard the pixelation and focus on the gradient difference on the color bars.
>What I'm getting right now is a full blown color bar with no difference at all

That is expected as it only supports 9 colours rather than the 16 (effectively 15) because there aren't enough comparators to distinguish the two bright levels of each colour in the YUV signal and it is would probably be too noisy anyway.

As mentioned in my previous post about this, the only way to get 16 colours cleanly is to use the vLA82S+ with RGB output or one of c0pperdragon's adapters:
https://github.com/hoglet67/RGBtoHDMI/issues/378#issuecomment-2118604160

However you should be getting two levels of white (plus one level of other colours) as shown in the gallery:
https://github.com/hoglet67/RGBtoHDMI/wiki/Gallery-of-Screencaps
and that doesn't look to be the case plus the white text is a bit distorted so you might have to tweak the Y DAC levels and maybe re-run calibration after that

>Trying now with an Issue 2, my original machine from 1982
>Currently I'm getting no sync detected

Some Issue 1 or 2 Boards may require links fitted to bring YUV to the edge connector, check the schematic.
(This is mentioned in the cables section.)

>These are the pinouts that "work for me"TM.

Can you post photos showing your wiring

luisfcorreia, 2024-07-26:
ok, I did have a small issue with wiring and I can confirm documentation is correct.
I'm getting YUV from very near the ULA (on the Issue 3B), will do Issue 2 after some rest.

As per 9 colors versus 15, I'm more then happy with it.

(I still have plans to tinker with FPGA/CPLD based designs)

Closing this as totally ok for now.

closed 2024-07-26 by luisfcorreia