hoglet67/RGBtoHDMI #394: Usable for old PC VGA output from 320x240 up to 1024x768?

8192K, 2024-08-09:
Apologies for opening an issue, but I don't seem to be able to ask a question anywhere else. If I missed something, please point me to it.

My use case it using an old DOS/Win98 based PC with an S3 Virge 325 graphics card and display the output on an HDMI monitor. This card produces resolutions from 320x240 (maybe 320x200, too?) to 1024x768 @16bit color depth. Anything higher is not required even though the card could manage 1600x1200.
Can I use this project to convert all resolutions in the given range, especially those DOS resolutions 320x240 and 720x400 @70Hz, to HDMI? I understand I have to assemble something, that would be acceptable. However, if there's something pre-assembled, I'd buy that, too. I'm in Germany if it matters.

Thank you!

IanSB, 2024-08-10:
No, VGA is not supported. This project is for earlier video standards such as EGA, CGA and 8 bit systems with 15Khz video (TV rate)

8192K, 2024-08-11:
Thank you. So it wouldn't even support a 320x240 resolution from a VGA card? 

Which solution would you suggest to get full VGA output for the mentioned resolution range on HDMI? Besides buying an old monitor, maybe.

braxlan, 2024-11-12:
> Thank you. So it wouldn't even support a 320x240 resolution from a VGA card?

No, _this_ project is meant for graphics cards which output TTL levels, VGA cards output an analog signal representing the RGB levels with different voltages.

btw, 320x240 will be pixel-doubled and line-doubled by the graphics card itself to meet the VGA timings, resulting in a 640x480 output (see [https://en.wikipedia.org/wiki/Video_Graphics_Array#Typical_uses_of_selected_modes](url))

> Which solution would you suggest to get full VGA output for the mentioned resolution range on HDMI? Besides buying an old monitor, maybe.

You could buy any VGA to HDMI converter. If you want to stay open source, I would recommend the _OSSC_ or the _OSSC Pro_. They both can also embed the audio coming from your computer(s) into the HDMI signal. Remember that the monitor you use has to support the input resolutions and refresh rates. The OSSCs support scaling, this helps getting around odd resolutions and timings a bit, but also adds some latency. You can also run them in "passthrough mode" and add an external scaler.

However, converting and upscaling open up a whole new chapter, but in my opinion it's absolutely worth it taking a deep dive.