hoglet67/RGBtoHDMI #399: Amiga 500 PAL - screen flashing in NTSC

mvbhz, 2024-08-16:
Games I start a game in NTSC the screen flashes cyclically every half a second - appears and disappears. How to fix it? In PAL games it is OK.

IanSB, 2024-08-16:
A flashing screen means that the video signal timing doesn't match any of the sub-profiles.
There are currently sub-profiles for:
PAL timing
NTSC timing
NTSC timing using a PAL crystal (when displaying NTSC on a PAL Amiga)
PAL timing using an NTSC crystal (when displaying PAL on an NTSC Amiga)

For some reason the signal you are outputting doesn't match any of those
Take a screencap of the source summary page in the info menu which should show the timing

mvbhz, 2025-01-02:
The image was jumping around a bit, now it doesn't appear at all in NTSC on my TV screen (message no signal). The TV definitely supports NTSC - I checked it on other NTSC equipment and the image is correct.

Here are the screenshots:
[image]
[image]

mvbhz, 2025-01-02:
> A flashing screen means that the video signal timing doesn't match any of the sub-profiles. There are currently sub-profiles for: PAL timing NTSC timing NTSC timing using a PAL crystal (when [...]

The image was jumping around a bit, now it doesn't appear at all in NTSC on my TV screen (message "Format not supported"). The TV definitely supports NTSC - I checked it on other NTSC equipment and the image is correct. I also tried different HDMI cables.

Here are the screenshots:

[image]
[image]

IanSB, 2025-01-02:
@mvbhz 
It looks like the correct profile is being selected and the timings are correct

>now it doesn't appear at all in NTSC on my TV screen (message "Format not supported").
I checked it on other NTSC equipment and the image is correct

Do you mean that you have tried the TV with other NTSC sources or you tried the RGBtoHDMI output on other monitors in NTSC mode and that works? (is that how you got the screencaps?)

What resolution is your TV?

The last two releases support auto detection of DVI or HDMI output mode depending on the monitor but I have seen HDMI signal quality problems with some monitors or TVs which work OK in DVI mode but are glitchy in HDMI mode and that can vary depending on the 50/60 refresh rate as well but it is usually on higher resolution monitors. 

Try going into the settings menu and changing HDMI mode detect to Manual and HDMI Manual mode from HDMI to DVI compatible. A reboot will be required to save the change. (Also try all the different HDMI options if that doesn't work)

mvbhz, 2025-01-02:
I connected another computer (e.g. THE64 American version) which has HDMI and transmits in the NTSC system. On the same TV which does not recognize the signal from RGB2HDMI (Amiga), the signal from THE64 is correct.

The native resolution of my TV is FHD (1080p, 1080i). The TV supports 50Hz, 60Hz. It also supports lower frequencies (e.g. 720p, 720i).

Tomorrow I will check on my second TV if it will recognize the image. I will also check the other settings you wrote about.

IanSB, 2025-01-02:
@mvbhz 

A few other suggestions if the above doesn't work:
Try changing the Genlock Mode to off in the settings menu as the refresh frequency being output is 59.28Hz instead of 60Hz due to the use of the non standard PAL crystal. Virtually all TVs should be able to cope with that slight difference but perhaps yours doesn't. Turning the genlock off should make it output the standard 60Hz frequency but you will likely get video tears or occasional frame drops instead with that setting disabled. 
Finally try changing the resolution from auto to something lower like 1280x720

mvbhz, 2025-01-03:
The image is displayed on another TV. It's probably what you wrote - the refresh rate is 59.28 but the first TV can't adjust to it. After turning off Genlock, the image in NTSC mode finally appeared on this TV.

Is it possible to set Genlock so that it is disabled but only for NTSC mode and for PAL so that it is enabled?

IanSB, 2025-01-03:
@mvbhz 
>Is it possible to set Genlock so that it is disabled but only for NTSC mode and for PAL so that it is enabled?

Yes, PAL and NTSC modes use different sub-profiles with their own settings which can be saved separately. However you have to make the change when already in that mode so you will have to connect to the working TV to do it.

Display NTSC mode, change the Genlock setting and then use the Save Configuration option in the main menu. That should only affect the NTSC sub-profile and when it auto-switches to the PAL sub-profile the genlock should be enabled. (The LED should be off for NTSC and on for PAL)

If you see video tearing effects with genlock disabled you can try changing the number of buffers from 1 to 2 or 3 in the settings menu which will change tearing effects into frame drops but that will also increase the lag by 1 or 2 frames.

IanSB, 2025-01-03:
@mvbhz 

I just realised that because any settings only affect the current sub-profile you would have to already be in NTSC mode to try any of the above suggestions like DVI compatibility which means you would first have to use a working monitor to make the changes and then save the configuration before switching to the TV for testing so you might want to try the other suggestions again in that way.
If you made the changes in PAL mode those changes would be lost when switching to NTSC and saving in PAL mode would only affect the PAL sub-profile.

mvbhz, 2025-01-05:
I turned off Genlock in NTSC mode and saved the settings. The image is there. Thanks for your help.