hoglet67/RGBtoHDMI #401: OHI Superboard II How To

sideburn, 2024-09-08:
I have an original Ohio Scientific Superboard II that has a composite video out and it will only sync to old CRTs. It has issues with composite put to a modern LCD so I would like to try the RGBtoHDMI on it. 

I see the Superboard II is on the list of supported machines under the YUV section. 

Are there any instructions available on how to connect this up to the board or could someone point me in the right direction?

IanSB, 2024-09-08:
@sideburn 

I think you already have a mono/lumacode board based on your previous post on c0pperdragon's github so you connect the video and ground input of that to the video & ground output of the superboard and select the superboard profile. (For the RGB analog board you connect the video to both the green and sync inputs)

Video is on pin 12 of J2 and GND is on pin 11 (J2 is the connector in the extreme rear left corner of the motherboard and pin 12 is the leftmost pin right in the corner)

Note that as the superboard video level is adjustable, you might have to make tweaks to get a stable image if it doesn't match the levels used on the default profile.

This can be done either by tweaking the DAC levels in the sampling menu or adjusting the pot on superboard by the left hand edge below J2.

sideburn, 2024-09-08:
Oh I didn’t realize I use the same ntsc composite video out that I am already using. I will give it a try. Thanks.

sideburn, 2024-09-08:
I can get it almost spot on. It just has random horizontal black lines. But it’s subtle enough to not be a show stopper. I’ve had issues getting ntsc composite video to sync to an lcd display. It would only sync to crt. I finally found and older lcd that would sync but it’s still a bit hit or miss. Maybe that’s related.

IanSB, 2024-09-08:
@sideburn 

That could be caused by power supply issues as I mentioned in the c0pperdragon thread but I recall I had to replace the video level pot on my superboard as it had become too dirty for reliable operation and was causing glitches (similar to a crackly volume pot on an old radio). You could also try cleaning it with deoxit but I tried that and eventually had to replace it anyway.

sideburn, 2024-09-08:
Well it’s running off battery power. It’s actually a portable / luggable conversion I did. You can see some pics here where it’s got the composite video lcd: https://osiweb.org/osiforum/viewtopic.php?t=882It looks like the lumacode boards are out of stock. I’ll wait til I get a new lcd panel with hdmi and a lumacode board and then tinker with the video out pot and see if I can improve it. The board I have now is dedicated to an Atari ST portable project. So far it looks much better than that old composite LCD so I’m going to dive into flat panel conversion once I can source the parts.

IanSB, 2024-09-08:
@sideburn 

If it's not the pot, then it is likely one or more of the DC to DC converters which introduces switching noise and is basically the same as using a mains switched mode PSU. That can upset old analog circuitry like the video output which was designed to run on linear supplies.
You could try a different DC to DC converter which might be less noisy but alternatively you could use a digital hook up using a standard digital RGBtoHDMI by picking up the separate video and sync signals at TTL levels which should provide noise immunity but that requires piggybacking a chip on the superboard to buffer those signals.

sideburn, 2024-09-09:
Ah I think I’m running into that right now o the Atari project. I move the power and I get noise. I think it’s the 5v step down converter and wondering if sharing the 5v with the logic board is causing issues. Wondering if it would be better to put the rgbtohdmi on its one step down board.