hoglet67/RGBtoHDMI #403: Color palette issues (lumacode)

sideburn, 2024-09-16:
I’m still having issues with this lumacode / rgbtohdmi on the PAL 65XE 

I’ve noticed now that the colors are off. When I bring up Atari Control Picture, the Y should be yellow and it is green. The B for blue is purple. And the R for red is orange. The bottom right corner of the color swatches should be yellow hues and they are green. 

I tried adjusting the the settings in color palette and dac settings sample menu but no luck yet. Any ideas ?
[image]

IanSB, 2024-09-16:
@sideburn 

Try adjusting the Pixel H Offset in the sampling menu
(Make sure any other changes you made to try fixing this have been undone)

sideburn, 2024-09-16:
I tried that as well. Photo attached of current settings. I have switched displays to a more modern one. Could it be something to do with the display not handling PAL maybe? I can switch back to the other monitor as a test.

sideburn, 2024-09-16:
Current settings :

[image]

IanSB, 2024-09-16:
@sideburn 

G Hi is far too high, it should be around 1.57v  + or - a relatively small amount, not 2.59v. G Mid and G Lo are lower than default but at least they seem plausible values. This means at least one of the 4 voltage levels isn't being detected properly.

I suggest restoring the profile and trying your adjustments again but limit all adjustments to small amounts although I think your underlying problem seems to be a very noisy power supply so getting clean power might be something else to try.

sideburn, 2024-09-16:
Well, I have tried multiple power supplies including a 5 amp battery. 
Do you mean to the logic board? I get the same results from the 65xe and the 130xe.
currently i am powering the RGBtoHDMI separately from a 2A usb power supply.
The logic board is powered by a separate 2A supply.

here is what the default settings look like:

￼

sideburn, 2024-09-16:
I may have to throw in the towel with this and try a Sophia or some other solution. I have spend days and hours fiddling with it. Changing out machines, power supplies and tweaking all of the params I can think of and the closest I could get was with those higher numbers. but so far no settings that I throw at it get the color palette any closer. Running out of ideas...

IanSB, 2024-09-16:
@sideburn 
I suppose it's possible that the RGBtoHDMI mono/luma board is faulty/damaged in some way that would cause it to have degraded performance or there is something wrong with the GTIA adapter board but as you were also having some noise issues with the superboard then maybe the former is more likely.

The only suggestion I have is to power the Atari from it's original linear PSU and use a different monitor with a built in USB hub, powering the Pi zero from that via a USB to micro USB cable as that will likely be a reasonably clean supply.  If there is still a problem then it may well be a bad board.

sideburn, 2024-09-16:
Yeah I think the board is bad… I have tried a 130Xe, 65xe, and superboard.. I have tried USB power supplies, wall wart supplies, and my bench power variable power supply all powered from same supply and each powered separately from seperate power supplies. I have also tried 3 different displays. One being an HDMI TV and two being flat panel monitors. I have exhausted all params and the 200/50 DAC settings are the only ones that get me remotely close and no matter what I get sprinkles of noise always in addition to the color palette issues I have now discovered.

eightbit0675, 2025-01-27:
I had an issue with wrong colors on the VIC-20 using the lumacode YUV profile. It wasn't until I used the profile named "unterm" that the colors were properly set. I have no idea why that is, but I am not complaining as now it works. Before the colors were all completely wrong and no amount of sampling adjustment could fix it.

When I set the unterm profile (what does that mean anyway?) it programmed the RGB2HDMI and rebooted and everything was perfect. 

NTSC VIC-20 btw.

IanSB, 2025-01-27:
@eightbit0675 

Unterm means unterminated i.e. 75ohm video termination is disabled. This has the effect of increasing the lumacode signal from about 1v peak to peak to 1.5v peak to peak which increases the noise margin so can be tried if there are problems with the standard terminated profile. It was included for just such situations as yours. (e.g. different PSUs on the VIC and the Pi might have different amounts of noise that affect the lumacode signal.)
Do you know what revision of VIC 20 you have and what issue and type of RGBtoHDMI you are using?

eightbit0675, 2025-01-28:
This is it with disabled video termination. Big difference :)

[image]

IanSB, 2025-01-28:
@eightbit0675 
>Well, I did have to increase saturation from 100 to 165 to get it to look like "how I remember", but the colors are correct and the image is perfect with this profile.

There will be an updated palette based on the vice emulator palette in the next beta release which should be more saturated 

>I asked c0pperdragon first and he thought it was weird that YUV did not work properly. It looked like this before:

That is strange, the expected problem would be a noisy image, not clean with wrong colours.

Which software version are you using?
Current Links here:
https://github.com/hoglet67/RGBtoHDMI/wiki/Latest-Software

Note if you install a new version you have to wipe the SD card to ensure all files get updated.

eightbit0675, 2025-01-28:
I've been using Beta 65 on this as well as my MCE2HDMI (Tandy 1000) with no issues (except for what I mention here in regard to the Vic-20). I went back to the Feb official release that rolled up everything up to Beta 64 according to the release notes, and it is the same thing.

I wiped the card each time before switching of course to ensure I have a fresh install, I am a little anal about doing that. I never copy over files ;)

Yeah, the image was pixel perfect, just completely wrong colors. Odd, I know. When I saw this thread with the OP having color issues I felt best to bring this up. I wonder if his problem is the same.

IanSB, 2025-01-28:
@eightbit0675 

I can actually reproduce those colours by changing the sampling mode from "6 bits (4 level)" to "6 bits per pixel" (which only detects three levels) in the sampling menu although how that is happening with your setup is unknown.
Do you use lumacode boards with any other computers and if so do the YUV profiles work OK?

eightbit0675, 2025-01-29:
I just found something I think. When I use the profile for YUV (and I am back to the strange colors) and I enter the sampling menu, it is set to "6  bits per pixel" and it cannot be changed. I cannot scroll through any other options. 

However, when I am set to "unterm" profile, it is set to "6 bits (4 level)" and the color is accurate, and I CAN scroll through the other options and choose the option that detects only three levels.

Maybe there is something wrong with the YUV profile? Why would it be stuck on 6 bits per pixel and no other option to change?

IanSB, 2025-01-29:
@eightbit0675 

When you are in YUV mode can you use the save log and edid option in the info menu and post log,txt

eightbit0675, 2025-01-29:
Attached!

[Log.txt](https://github.com/user-attachments/files/18588925/Log.txt)

IanSB, 2025-01-29:
What type and issue of RGBtoHDMI board are you using? (i.e. mono/lumacode or standard CPLD with analog board) because the log indicates that it thinks it is running on a 6 bit CPLD board which is a very old design:
```
Pi Hardware detected as type 1
CPLD board detected
6 bit board detected
Standard board detected
```
That would explain the problem because 4 level analog used by lumacode is not supported on that old design which is why you can't select 4 level in the menus.

One possible cause for this is that R7 is missing, faulty or not soldered properly (R7 is a 1K resistor near the reset button)

eightbit0675, 2025-01-29:
I'd have to open this to find out. I didn't build it. There is a person on Tindie selling these pre-built. But, before I do that I will email him and see what he says.

eightbit0675, 2025-01-29:
This is the link to the sale page:

https://www.tindie.com/products/rdupuy5000/rgbtohdmi-hdmi-adapter-analog-version-complete/

Says its:

RGBtoHDMI CPLD board
RGBtoHDMI Analog Board

eightbit0675, 2025-01-29:
In the meantime I opened it (I was afraid of cracking the 3D print, but it was OK!)

Looks like R7 missing?

[image]

eightbit0675, 2025-01-29:
The person on Tindie has reached out and admitted to the error. Very nice person and is sending me a replacement hat with R7 in place.

I appreciate your help Ian, I would have never discovered this without it!