hoglet67/RGBtoHDMI #404: Color is lost and displayed in B/W

clumpwork, 2024-09-16:
I turned on my Apple IIe for the first time in almost a year and noticed that it could no longer display colors. The display is in black and white. Text and graphics are still displayed, but the colors are completely lost. The cable connection is fine.
I have also confirmed that there is no problem with Raspberry Pi Zero. Could this symptom be the result of a CPLD malfunction?
I would like to investigate the cause of the problem, but any advice would be appreciated.

IanSB, 2024-09-16:
Try adjusting the Y Lo DAC value in the sampling menu as suggested in the info menu under "Help NTSC Artifacts"
Also you can try the artifact colour manual override by holding SW2 for a couple of seconds (with no menu on screen) to toggle colour artifacts on/off.

clumpwork, 2024-09-17:
Thank you for your answer.
NTSC artifact detection is working. I have tried toggling color artifacts on and off as you suggested, but the symptoms remain the same. I have tried rebooting the CPLD and updating the software to the latest version, with the same result.

IanSB, 2024-09-17:
@clumpwork 
Which profile have you selected?
Are you using an analog board via the phono output or a direct digital pickup?
When you do a long press of SW2 do you get the "NTSC color on" or "NTSC color off" message displayed?
Can you post a screenshot of the showing the graphics (that should be in colour) after "NTSC color on" has been displayed.
Also it's probably worth using the "Save Log and EDID" option in the info menu and posting log.txt from the SD card

clumpwork, 2024-09-18:
Thanks for your comment.
The profile I selected is “Apple IIe TTL”. I used “A direct digital pickup” for the connection, and when I pressed and held SW2, the message “NTSC color on/off” was displayed.
Attached is a screenshot of the graphics are displayed in B/W instead of color and log.txt file .
What I don't understand is that the graphics suddenly changed to B&W, even though they were in the same state as before. I may need to borrow a friend's RGBtoHDMI to see if it is reproducible.
[image]
[Log.txt](https://github.com/user-attachments/files/17046403/Log.txt)

IanSB, 2024-09-18:
@clumpwork 

OK it looks like the colour switching signal (IOU GR) which is connected to Green2 (pin 4 on the 12 way CPLD header) is not working. This could either be due to the wire being disconnected inside the Apple IIe or maybe that input had a bad solder joint on pin 36 of the CPLD or the CPLD is damaged in some way.

clumpwork, 2024-09-19:
Thank you for showing me your detailed analysis.
I have checked the signal from the Apple IIe and there is no problem. There is also no problem with the RGBTOHDMI board, and the signal goes through to pin 36 of the CPLD.
Then the cause of this symptom is probably the last one, the damage of the CPLD.
I will try to replace the CPLD.

clumpwork, 2024-09-29:
I got a new CPLD and replaced the damaged one. The problem was solved. I can now display in color. Thank you!