hoglet67/RGBtoHDMI #405: Sega Megadrive 2 cabling

G4DDS, 2024-10-06:
Have been given Mega Drive 2 and tried using a SCART lead (with another converter) but would prefer a crystal clear HD picture. From the cables section I see there are no consoles at all, but surely its just as a matter of wiring up the correct lead? The video pin outs have RGB, GND and CSYNC so is it just the same setup as an one of the existing Analog setups (like Amstrad or Spectrum +3):

Pin 1 GND
Pin 2 SYNC
Pin 3 BLUE 
Pin 4 GREEN
Pin 5 RED
Pin 6 +5V

G4DDS, 2024-10-11:
I've had a go but not making much progress. I have followed the Amstrad wiring and selected that as a profile. I kept getting a blank white screen but I discovered changing the DC coupling to AC (which I believe should only be used with YUV and not RGB) I get some sort of picture:

[image]

I don't want to spend too much time and effort if I am going down the wrong path with this. Is the Amstrad wiring suitable? Is there another base profile that maybe better to work with?

IanSB, 2024-10-11:
@G4DDS 

This isn't really going to work well because the analog board can only discriminate 4 levels each of RGB but the palette of the megadrive has 8 levels each of RGB. With the correct profile you would get a reasonable image but colours might be slightly inaccurate or flickery.

Anyway, here are a couple of base profiles to try:

[NTSC_4LVL_Base.txt](https://github.com/user-attachments/files/17344911/NTSC_4LVL_Base.txt)
[PAL_4LVL_Base.txt](https://github.com/user-attachments/files/17344912/PAL_4LVL_Base.txt)

Put these in \Profiles\6-12_BIT_RGB_Analog\_Base_Profiles and select the PAL or NTSC one as appropriate
These have 4 level palettes selected but you will probably have to adjust the DAC levels for lo, mid and hi depending on the analog video levels output by the megadrive and you will also have to use the Create Custom Profile menu and Analyse Timing etc to get the settings right.

It looks like the video is analog RGB out of the video chip so it wouldn't be possible to pick up the digital bits directly like you can with some systems so the only other alternative would be a c0pperdragon style lumacode board if one was developed.

c0pperdragon, 2024-10-14:
Sorry, but the Mega Drive is just out of reach of my possiblities and the abilities of the lumacode system.

G4DDS, 2025-03-30:
I'll close this issue as I know have an alternative to the RGBotHDMI for the Mega Drive.

closed 2025-03-30 by G4DDS