hoglet67/RGBtoHDMI #407: CPLD Recovery Menu - Confirm programming in single button mode not possible

hudecitydave, 2024-11-14:
Hi there!

I have my litte solderjob done (LinixJedi Version) here and got my CPLD Recovery menu presented now. So far so good. I copied the config files and profile to their positions, and I also checked single_button_mode=1 in Default. So should be happy.

But...

I am able to surf the whole menu behind "return", can set settings short-klicking to select things and long clicking to confirm and so on. working fine.

But only on the point --Current: BLANK: Confirm?--

I get stuck.

If I short click, I leave the selection, and go from RGB Item to next YUV Item (as expected).

If I long click to accept programming, while holding button status flashes between "Confirm" and "Select Correct CPLD".
Feels like it does not take a long click for OK, instead it fires instant shorts in a row or something.

Again - the whole one-button function is working well, only the one point confirming the programming get stuck.

What am I doing wrong here?

Thank you so much!

IanSB, 2024-11-15:
@hudecitydave 
> If I long click to accept programming, while holding button status flashes between "Confirm" and "Select Correct CPLD".
> Feels like it does not take a long click for OK, instead it fires instant [...]

I just tried putting one of my test boards into single button mode and erasing the CPLD to BLANK. Then I put a clean copy of the latest stable release on the SD card and that all programmed successfully using a long press to select the CPLD and a second long press to confirm so it does seem to be working OK.

If you got your CPLD from a unofficial source like ebay or China then it is possible that it already has some other firmware installed and that can stop initial programming from working.

Try programming again then after it fails go to the info menu and use the save log and EDID option and post log,txt as that might show a programming error.

For standard boards, the workaround for this is to temporarily cut the CPLD recovery jumpers on the underside for initial programming (you must remake them after initial programming) but I don't think linuxjedi included those on his board which makes the workaround a bit harder as you have to lift some pins on the CPLD.

hudecitydave, 2024-11-16:
Thank you. I think the chip is bad or something went wrong. I try the log option or start over.

closed 2024-11-16 by hudecitydave