hoglet67/RGBtoHDMI #410: Slightly different wiring of a Spectrum 128+ for Analog board

c0pperdragon, 2024-12-12:
The Spectrum 128+ is the only Spectrum that would require a digital RGBtoHDMI. All the others will work with the analog variant. With the lumacode modification, even the 16K/48K variants can use this and show all 16 colors.
You have provided instructions how to combine R,G,B and I to make it somhow work, but this hack will basically cause the ULA pins to be shorted together, which is not so great in my opinion. 

Maybe a better way would be to just combine G and I with 1K and 2K resistors to create 4 voltage levels. From this, the state of the I output could always be determined, regardless of the state of the color signals. Maybe this could even be useful to show the gray color if the user wants this. 
To stay in the allowed analog range for all 4 combinations, an additional 1K resistor to GND is also needed. 

I could build these resistors into the cable, and it would be easier than with diodes because their orientation would not matter.  

Can you think of any reason, why this should not work?

IanSB, 2024-12-18:
@c0pperdragon 
The ULA pins aren't directly shorted together as there are already some resistors in series with the RGBI out on the 128. A similar diode circuit is implemented in the +2A. Your suggestion might work but I don't have that particular hardware to do any tests.

c0pperdragon, 2024-12-18:
So if the wiring with the diodes is the true and tested variant, I will provide cables that are wired like that.

closed 2024-12-18 by c0pperdragon