hoglet67/RGBtoHDMI #411: Flashing screen w/ Tandy Model 1 Mod

snhirsch, 2024-12-12:
I've tried everything suggested in the troubleshooting doc, but to no avail.  When the onscreen menu is displayed, the output from the computer is solid and stable.  As soon as I dismiss the menu it starts flashing about 2x per second.  The computer is modified per your documentation to bring out the pre-analog H sync.

Has there been a regression on this particular machine?  I have tried both the most recent release and Beta 57.  No change.

IanSB, 2024-12-18:
@snhirsch 

The NTSC version had the wrong settings for number of lines in the profile. This was fixed earlier this year but I haven't released a new beta with that fix yet.

Here are the corrected 60Hz sub-profiles for the standard and modified profiles:
[TRS_80_Mod_60Hz.zip](https://github.com/user-attachments/files/18186547/TRS_80_Mod_60Hz.zip)
Unzip and overwrite those files in the appropriate sub-folders in \Profiles\6-12_BIT_YUV_Analog\Tandy_+_Dragon\
(One in each folder)

snhirsch, 2024-12-18:
Thanks!  Looks a lot better now.

closed 2024-12-18 by snhirsch