hoglet67/RGBtoHDMI #414: Can this work with Atari TT?

mzryz, 2025-01-08:
Hi all,

The Atari TT had VGA output but it had slightly different timings to the official standard, this causes the TT output to any regular LCD to be a bit iffy.

VGA PixelClock is 25MHz and in case of the TT PixelClock it is 32MHz (for color modes ) because it's clocked by the TT mainclock.

An OSSC can be used to adjust the image and get pixel perfect output, but this is an expensive option. 

Would the RGBtoHDMI be powerful enough to process this kind of signal? If it is, perhaps I can think about trying it out.

Cheers

IanSB, 2025-01-10:
@mzryz 
Looking at the schematic, the TT Video chip has analog RGB out rather than digital RGB like the other shifter chips so it would not work.

mzryz, 2025-01-11:
Thanks for looking into this Ian, much appreciated

closed 2025-01-11 by mzryz