hoglet67/RGBtoHDMI #415: C64 lumacode stopped working in the latest beta release

rjpontefract, 2025-01-23:
I was attempting to get my lumacode RGBtoHDMI working with a ZX Spectrum using C0pperdragon's ULA digitizer.  It is currently being used with a C64 using firmware version 20230517_eb620884 and is working fine.  However, it seems that I needed to upgrade to version 20240213_e630d36f to get a Spectrum 48K lumacode profile.  I had problems getting the Spectrum output to be stable, so I reattached the C64 and got a blank screen after changing the profile back to C64 lumacode.

I restored firmware version 20230517_eb620884 and the C64 worked again as before.

Is there something I'm missing when I upgraded the firmware that would cause it to not work with the C64?

Any suggestions would be appreciated.

IanSB, 2025-01-23:
@rjpontefract 
I'm not aware of any issue with the C64 using the latest stable release.
Is your C64 PAL or NTSC?
What profile are you using? (i.e. the RGB (unterm) or the YUV one)

Could you also try the latest beta here:
https://github.com/IanSB/RGBtoHDMI/releases

rjpontefract, 2025-01-23:
Thanks for replying.  It's a PAL C64 and I'm using the C64 Lumacode profile.

I tried the latest beta that you linked to and it's working again.  I suspect user error was involved somewhere, but I'm happy to be back up and running.

rjpontefract, 2025-01-24:
Now that I have the C64 backup and working, I'd like to ask for a little help with the 48K Spectrum please.

It has a C0pperdragon ULA digitizer installed and, if I plug that directly into the composite input of the monitor, I get the expected lumacode output.

I plugged it into the RGBtoHDMI that I use with the C64 and changed the profile to:

[image]

This gave an unstable, but clear view of the spectrum display.  The display flickers and occasionally I see the No Sync Detected error.

The source summary screen shows:

[image]

And the system summary screens shows:

[image]

I did read the online help for flashing screens, and I tried adjusting some of the recommended parameters, one at a time, with no luck.  Could you provide any suggestions for what I may be doing wrong, or what adjustments may help?

IanSB, 2025-01-24:
@rjpontefract 
First thing: when you upgraded the software did you format or wipe the SD card as you can't upgrade by simply overwriting the existing installation.
Have you tried the terminated Lumacode profile (The YUV version without the "(unterm)")
If you are re-using the modulator phono, did you make sure the existing signal was disconnected from the phono centre pin.  
Can you post a photo of the direct to monitor lumacode output

rjpontefract, 2025-01-25:
Thanks again for replying.

I did wipe the SD card before upgrading, but that may have been my mistake earlier with the C64 not working.  When I upgraded to the latest beta I made sure to erase the SD card first, which I didn't do when I upgraded from 20230517_eb620884  to 20240213_e630d36f.  Thanks, that explains why the C64 stopped working.

I have tried both terminated and unterminated profiles and there was no difference that I could see.

I'm not currently using the modulator phono, just the one attached the the ULA digitizer board.  I'll heed your warning if I decide to go that route when I reassemble it.

Here's a photo of the direct to monitor lumacode output:

[image]

However, I'm glad that you asked me to do that, as shortly after that picture was taken, I started seeing sync errors on that output as well.  Given that it was over 30°C in the room I figured maybe it was heat related, so I applied some freezer spray to the ULA and the ULA digitizer board, and what do you know, I got the following output from the RGBtoHDMI:

[image]

So it seems like something sync related drifts when it gets too warm.  I'm not sure what to do about that other than to wait until Winter comes to play with the Spectrum.  Any suggestions would be appreciated.

Thanks again for your help on this, it's appreciated, as is the whole RGBtoHDMI project.

IanSB, 2025-01-25:
@rjpontefract 
>I made sure to erase the SD card first, which I didn't do when I upgraded from 20230517_eb620884 to 20240213_e630d36f. Thanks, that explains why the C64 stopped working.

That was likely the problem.

As the C64 works and the Spectrum problem seems to be heat related this doesn't seem to be a problem with the RGBtoHDMI side of things so you are probably better off raising an issue on c0pperdragon's github in the Spectrum thread:
https://github.com/c0pperdragon/LumaCode/issues

rjpontefract, 2025-01-26:
I agree, and thanks very much for your help.